julianschill/klipper-led_effect #1: Doesn't seem to work

aepkolol, 2021-11-21:
Installed via the script,  using the "breathing panel_idle" lines in the config doc, I get an error from klipper 

```
Internal error during ready callback: could not convert string to float: [(.5
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
```

If I try and make my own code, addressing arrays of LEDs hooked to my printer, I get index out of range errors.

julianschill, 2021-11-21:
Sorry, this is actually not ready for use yet. Not all effects are working and the documentation is not up to date. This repo is mainly for collaborating and getting it done. I will add a note to the Readme.

I checked the documentation and found that the example you picked has still an old format.
So if you want to try it out anyway, remove the square brackets and maybe increase the effect rate in the config:

`breathing  10 1 top (.5,.5,1)`

julianschill, 2021-12-15:
Fixed in documentation

closed 2021-12-15 by julianschill