julianschill/klipper-led_effect #6: Where to place "color_order_GRB: true" statement

Markus-1984, 2021-11-24:
Hi,

first I want to say that you did great work.
Unfortuanly I don`t know where to put in the "color_order_GRB: true" statement.
I´ve tried in the config of my LED, in the config of my LED-Effect and in the config of the LED-Effect in the leds: section.
Please give me a hint.

Best regards
Markus

julianschill, 2021-11-24:
Documentation is outdated in this regard. Put this in your neopixel section:

```color_order: GRB```

Markus-1984, 2021-11-24:
That's what I've done in the config of the LED. If I set the color via Gcode, it's workin well. 
But in the effects, the palette is still in GRB instead of RGB. Is it correctly?

julianschill, 2021-11-30:
I checked and it seems to be a bug. Thanks for pointing it out.

julianschill, 2021-12-12:
@Markus-1984 I pushed a change that should fix this issue. Can you please test if it is working correctly now?

Markus-1984, 2022-01-23:
Hi julian,

now it's working like it should. Thanks.

closed 2022-01-23 by Markus-1984