julianschill/klipper-led_effect #12: Fire and heater_fire bug

BrokenGhostWest, 2021-12-18:
The heater and heater fire animations when set to cutoff 1 do not stop working when the heater cools, they just stop displaying. For example if you use the 2 config examples and autostart bed efects and pannel idel the machine will breathe as it powers on, display fire untill temp and shutoff. pn cooling it displays back down untill min temp but then it stop breathe starting and stays on the first colour of the fire animation list, eg red.

julianschill, 2022-01-24:
I think this has to do with the current behaviour when there are multiple effects running on the same LED. Then the last one (which is probably the lowest one in the config) wins and overwrites previous effects. With cutoff=1 the heater effect basically puts out a 0 value. It think #3 will fixes this issue. Until then run `SET_LED_EFFECT EFFECT=your_heater_effect STOP=1` to disable the heater effect, when it should not be shown.

julianschill, 2022-01-24:
It is not 100% clear to me, though. Can you maybe provide an example config to reproduce the behaviour?

julianschill, 2022-02-23:
@BrokenGhostWest is this issue still pending or can I close it?

julianschill, 2022-03-05:
Should be fixed by #33

closed 2022-03-05 by julianschill