julianschill/klipper-led_effect #13: led_effects is borked for use with dotstar led chains

TodWulff, 2021-12-27:
As requested, this is your reminder... ;-)

"**Hagbard V2.1070**  Thank you. I think I know what the issue is. I made a change to read the color_order from the config, and apparently for dotstar there isnt such a  configuration. So for dotstar I have to assume the default color order. Should be a quick fix, once I'm home. Would you mind creating an issue on github as a reminder for me?"

Thread start: https://discord.com/channels/460117602945990666/635687829254701107/925020306156970024

TodWulff, 2022-01-22:
Good day.  Would it be useful to provide you with a short strip of DotStart LEDs?  If so, PM me in Discord with your address and preferred carrier when shipping from the US and I will get some coming your way.  I understand your in the Czech Republic.

julianschill, 2022-01-24:
Fixed with 2e84f6b

closed 2022-01-24 by julianschill