julianschill/klipper-led_effect #19: RGB to RGBW color conversion with color temperature

julianschill, 2022-01-06:
The RGB colors are converted to RGBW values, when a RGBW strip is connected. This also means, that a color temperature has to be configured.

Currently the color temperature of the white led is hardcoded. This has to be moved to the configuration.

drelich, 2022-09-22:
Hi, is there any update on this one? Really love your plugin but having a hard time setting the white colour on my RGBW neopixel strip. Thanks.

julianschill, 2022-09-25:
I am working on it in the [rgbw_support #](https://github.com/julianschill/klipper-led_effect/tree/rgbw-support) branch. If you like you can test it. Or just wait.

drelich, 2022-09-25:
I'll give it a try and report back with results :) Thanks!

julianschill, 2022-09-25:
In this branch you are able to set colors using 4 values like this: (1.0, 0.4, 0.3, 0.2). If you set only 3, the white channel will be 0.

drelich, 2022-09-25:
Interesting. Grabbed the rgbw-support branch and installed it, still getting dark red light when I set it to (0,0,0,1). Must be overlooking something. My neopixels are set to GRBW so that should be fine and they work using the Klipper's built in SET_LED...

julianschill, 2022-09-25:
@drelich can you please post the effect configuration, you are trying it with?

drelich, 2022-09-25:
```
[led_effect nozzle_dimmed]
autostart: false
frame_rate: 24
leds:
    neopixel:toolhead (2,3)
layers:
    static 1 1 top (0,0,0,0.3)

[led_effect nozzle_on]
autostart: false
frame_rate: 24
leds:
    neopixel:toolhead (2,3)
layers:
    static 1 1 top (0,0,0,1)
```

They're basically two presets to set the nozzle LEDs to dimmed white or bright white, nothing fancy. 

Checked the branch I'm on and it says rgbw-support, ran the install-led_effect.sh after switching to that branch (and MainSail does complain about not being on the master branch for updates).

julianschill, 2022-09-25:
I tried your example and the two LEDs light up white as they should. Do you by chance have other LEDs to test it with? Is your klipper up to date? Also make sure to restart klipper or even better reboot.

drelich, 2022-09-25:
Those are the only ones I have. Everything's updated to latest, incl. Klipper. Also rebooted several times. Those LEDs work fine with Klipper's SET_LED.

Is it possible I need to uninstall the master version and then re-install the rgbw-support branch? If so, how would you recommend I go about that? Thanks.

julianschill, 2022-09-25:
Ok, I just tried with the master branch and one LED lights up red. What you can do is delete ~/klipper/klippy/extras/led_effect.py and run the installation script again. Maybe you copied it instead of linking it.

drelich, 2022-09-25:
Ha! Okay, I have not removed that one. Completely wiped that file, along with all the other downloaded files, did a clean install from the rgbw-support branch and now it WORKS! :) Thank you!

julianschill, 2022-10-06:
fixed by #93

PR #90 (2022-09-25): RGBW support
Palette now supports 4 values for colors. Removed RGB to RGBW conversion. Fixes #19
closed 2022-10-06 by julianschill