julianschill/klipper-led_effect #20: SET_LED_EFFECT only works once per effect

Sly-Ripper, 2022-01-09:
It seems that SET_LED_EFFECT will only trigger an effect once.

Example
`SET_LED_EFFECT EFFECT=a` works
`SET_LED_EFFECT EFFECT=b` works
`SET_LED_EFFECT EFFECT=a` stays on 'b'

Gilliearnie, 2022-01-14:
I recognized the same issue.
I am starting with an idle effect, if I choose now a new effect, both effects will be used.

It is getting strange, if I am using thre or more effects by typing `SET_LED_EFFECT EFFECT=`
I tried to use an "off" command like defining

```
[led_effect off]
autostart:              false
frame_rate:             24
leds:
    neopixel:statuspixel
layers:
    static         1  0     top        (0.0, 0.0, 0.0) 
```

But it doesn't switch off the LEDs.

julianschill, 2022-01-14:
At the moment effects are not merged, but the second effect overwrites the previous one on the same LED.
Effects run until they are stopped. Use `SET_LED_EFFECT EFFECT=a STOP=1` to stop a specific effect and `STOP_LED_EFFECTS` to stop all effects.

julianschill, 2022-01-24:
Fixed by 3eaf588

closed 2022-01-24 by julianschill