julianschill/klipper-led_effect #28: SET_LED after gets overwritten after stopping LED effects

julianschill, 2022-02-08:
So my printer sits idle with two effects running, a rainbow gradient for case lights and one for stealthburner leds. In my print_start I call SET_LED_EFFECT EFFECT=party STOP=1 and SET_LED_EFFECT EFFECT=party2 STOP=1 to stop that effect and then I call a standard klipper set_led command to set the case lights to full white, and then the SB macro for STATUS_HEATING. 

Sometimes this all works correctly, most times what happens is the effects stop, the case lights do NOT set to white and the SB does set to status heating, sometimes neither LED strip gets set and they're just off after the effects turn off.

If I do it manually it's fine, if I put the set led calls later in the Print_start macro it's fine, it seems like a timing issue of some sort, like they're being executed too close together and something doesn't work right

burntcoinpurse, 2022-07-14:
You can try to add

SAVE_GCODE_STATE NAME="a name"
Your gcode macros, then 
RESTORE_GCODE_STATE NAME="a name"

That tends to help mine

MaffooClock, 2022-11-08:
_After writing all of the below, I just now noticed this issue was opened by the repo owner.  Thus, as he's already aware of this, my comment isn't really helping anything, except maybe to contribue some elaboration, for whatever that may be worth._

-----

I like to have a default LED effect on my Stealthburner (I have the Rainbow Barf in the logo, it's fun).  I also use [these macros](https://github.com/sttts/voron-klipper-config/blob/master/stealthburner_leds.cfg#L33-L46) to automagically set the logo and nozzle LEDs based on certain states/activities (e.g. `STATUS_BUSY` turns the logo LEDs red).

I've modified those macros to include `STOP_LED_EFFECTS LEDS="neopixel:sb_leds (1-8)"` as the first command, but most times it doesn't work; the LEDs continue with the effect, or sometimes the effect gets stuck with weird colors.  Sometimes if I repeat the `STATUS_*` macro two, three, seven times, it'll eventually work, but most of the time it doesn't.

But like you said, if I run the contents of the macro manually in the console, it works fine.

Other things I've tried (for posterity):
1. adding the `STOP_LED_EFFECTS ...` multiple times at the start of those macros, doesn't seem to help.
2. adding `G4 ...` (dwell) commands to induce various delays.
3. @burntcoinpurse's suggestion to use `[SAVE|RESTORE]_GCODE_STATE`

I really dig this module, and may find myself digging into the source more to see if I can help.

MaffooClock, 2022-11-08:
> Try STOP_LED_EFFECTS

I haven't tried that as I have three neopixel groups across two RGB channels, and I didn't have a reason to stop everything everywhere.  But for testing purposes, may as well give it a shot and See What Happens™

> If you're trying to run multiple effects at different times on 1 strip...

I'm trying to keep things simple while setting this up, so there's only three effects defined, each one running on only one of the three neopixel groups.

> Also if they are on delays or fadetimes, you have to let that time run out before you can issue more commands.

Good point.  No delays or fades in my config, but good to know.

Thanks for the input!

MaffooClock, 2022-11-08:
> I fixed this already

Orly?   That must be commit 1feb95e.  I thought I was using develop branch, but apparently I overlooked something on my end.

Thanks brotha, sorry for the spam!

julianschill, 2022-11-08:
If you are in fact running the develop branch and still have the issue, let me know, please.

MaffooClock, 2022-11-09:
I certainly remember running `git checkout develop`, but apparently I never ran the `install-led_effect.sh` script.  So, yeah, confirmed: I am two kinds of idiot :)

It works perfectly.

Thanks, and my apologies for the unnecessary traffic.

PR #101 (2022-10-23): Optimize enabling and disabling so it executes immediately.
Fixes #28
commit 1feb95e8 (2022-10-23): Optimize enabling and disabling so it executes immediately. (#101)

Fixes #28