julianschill/klipper-led_effect #29: [Feature Request] Toggle or gcode command as toggle for use with gcode_button

exiom-hk, 2022-02-10:
I suppose what I have in mind is a feature request.

Currently,
>Defined effect called panel_idle that can be controlled via the gcode command SET_LED_EFFECT EFFECT=panel_idle. Calling the command enables this effect. It can be disabled again by calling SET_LED_EFFECT EFFECT=panel_idle STOP=1`

What I would like to propose is perhaps when running a command that is already in effect, turns it off, instead of requiring `STOP=1`

Or perhaps the author has a different approach in mind where one could toggle a particular set of LEDs (enclosure lights for example) on/off via a physical gcode_button.

I understand not many people have physical buttons so I'll understand if it is not considered.

julianschill, 2022-02-10:
I won't implement toggle as default. But maybe I could add a command to query the current state, if necessary. But for your use case it is not. You can use macros with a variable to solve this:
```[gcode_macro lights_on]
gcode:
  SET_LED_EFFECT EFFECT=my_effect
  SET_GCODE_VARIABLE MACRO=toggle_lights VARIABLE=state VALUE=1

[gcode_macro lights_off]
gcode:
  SET_LED_EFFECT EFFECT=my_effect STOP=1
  SET_GCODE_VARIABLE MACRO=toggle_lights VARIABLE=state VALUE=0
  
[gcode_macro toggle_lights]
variable_state: 0
gcode:
  {% if printer["gcode_macro toggle_lights"].state == 0 %}
    lights_on
  {% else %}
    lights_off
  {% endif %}```
  
Then just call the toggle_lights macro from your button. If you make sure to only use the macros, instead calling SET_LED_EFFECT on that effect directly, the state is always tracked.

exiom-hk, 2022-02-10:
Ah thanks so much, I did not know you can do this... this example will help me tons now that I know this is possible

closed 2022-02-16 by julianschill