julianschill/klipper-led_effect #31: critical_error does not work

intositeme, 2022-02-15:
Nice work. Though the docs is a little hard to understand.

I can't seem to get critical_error to work, even when i get errors mid print or mcu shutsdown.
I have it set to
```
[led_effect critical_error]
leds:
    neopixel:case_light (1, 9, 10, 18)
layers:
    strobe         1  1.5   add        (1.0,  1.0, 1.0)
    breathing      2  0     difference (0.95, 0.0, 0.0)
    static         1  0     top        (1.0,  0.0, 0.0)
autostart:                             false
frame_rate:                            30
run_on_error:                          true
```

NameOfTheDragon, 2022-03-09:
I think this might be a Klipper issue - I seem to remember reading a conversation between Paul and Kevin where they were discussing this issue, and I think Paul had to resolve it in his private fork of Klipper. I will try to find the conversation.

julianschill, 2022-04-24:
I found it here: https://github.com/Klipper3d/klipper/issues/2725

Will look into it if I have some time. 
FWIW I fixed an exception, that occured when Klipper was going in shutdown.

julianschill, 2022-09-25:
This needs patched MCU firmware. So quite hard to do in a klipper plugin

smwoodward, 2023-06-01:
> This needs patched MCU firmware. So quite hard to do in a klipper plugin

Is there any update on where to find or make the changes to patch the MCU firmware for this to work?

julianschill, 2024-04-28:
That's the patch for the Klipper firmware:
```
diff --git a/src/neopixel.c b/src/neopixel.c
index bbea09f8..f2da3a20 100644
--- a/src/neopixel.c
+++ b/src/neopixel.c
@@ -186,7 +186,7 @@ command_neopixel_update(uint32_t *args)
         shutdown("Invalid neopixel update command");
     memcpy(&n->data[pos], data, data_len);
 }
-DECL_COMMAND(command_neopixel_update,
+DECL_COMMAND_FLAGS(command_neopixel_update, HF_IN_SHUTDOWN,
              "neopixel_update oid=%c pos=%hu data=%*s");
 
 void
@@ -197,4 +197,4 @@ command_neopixel_send(uint32_t *args)
     int ret = send_data(n);
     sendf("neopixel_result oid=%c success=%c", oid, ret ? 0 : 1);
 }
-DECL_COMMAND(command_neopixel_send, "neopixel_send oid=%c"); ```

And check this branch for the changes in led_effect: https://github.com/julianschill/klipper-led_effect/tree/julianschill/issue31

andreykurichev, 2025-06-28:
Извините за беспокойство, я разобрался :)
 
Краткая инструкция для тех кто также как и я делает это.
1. В файле /home/pi/klipper/src/neopixel.cв самом низу файла в функции void command_neopixel_update(uint32_t *args)

DECL_COMMAND(command_neopixel_update

заменить на

DECL_COMMAND_FLAGS(command_neopixel_update, HF_IN_SHUTDOWN,
              "neopixel_update oid=%c pos=%hu data=%*s");

DECL_COMMAND(command_neopixel_send, "neopixel_send oid=%c");

Заменить на

DECL_COMMAND_FLAGS(command_neopixel_send, HF_IN_SHUTDOWN, "neopixel_send oid=%c");

Oстановите klipper.
Скомпелировать прошивку и загрузить в MCU.

2. Измените led_effect.py
Если у вас новая версия Klipper в моем случае v0.13.0-154
используйте файл led_effect.py из этой ветки. [ ](https://github.com/julianschill/klipper-led_effect/tree/julianschill/issue31)

измените for i in range(ledChain.led_helper.get_led_count()): 393 строка
на for i in range(ledChain.led_helper.led_count):

И все должно заработать)))