julianschill/klipper-led_effect #36: heater layer turns back on when disabling heater

hapklaar, 2022-02-26:
First of all thanks for making this and making it available to us, really cool project!

I made the following config to light up my led strip in red, when the extruder heater is trying to reach a target temperature and be white when it is idle or reached its temperature:

```
[led_effect tool_effects]
autostart: true
frame_rate: 24
heater: extruder
leds:
    neopixel:leds
layers:
    heater  50 1 top    (1,0,0),(1,1,1)
    static  0  0 add    (1,1,1)
```

This works, but it also turn the heater layer back on when the extruder heater is switched off, resulting in the led strip also becoming red when it is cooling down before it reaches 50. Is this by design and if so how should I configure this so that it only turns red when the extruder is heating up?

hapklaar, 2022-02-28:
This seems to be governed by the first number after heater. I changed this to heater 300 which causes the effect to stop below 300 when cooling.

Still not sure how the palette works for heater. For me it is always red and doesn't change with temperature.

julianschill, 2022-03-01:
Current implementation is like this:
When you switch the heater on it will go through a gradient of the palette, except the last value until the temperature is reached and will then switch to the last color of the palette.
Example: This will go from blue to red and switch to white, when temperature is reached:

```[led_effect heating_test]
leds:                               
    neopixel:leds
autostart: true
frame_rate: 24
heater: extruder

layers:
    heater  50 0 top    (0,0,1),(1,0,0),(1,1,1)
```

If you want the effect to show go black instead of the last color set the second parameter to 1:
`heater  50 1 top    (0,0,1),(1,0,0),(1,1,1)`

When the heater gets switched off and the heater cools down, it again goes through the gradient until the temperature defined in the first parameter (50°) is reached and then turn off (effect outputs 0,0,0).
If you set the parameter to 0, the effect shows (0,0,0) when cooling down (off):
`heater  0 1 top    (0,0,1),(1,0,0),(1,1,1)`

I hope that clarifies and solves your issues. 
@hapklaar: Set first parameter to 0 if you do not want the effect to be active while cooling.
@hapklaar and @mhegering: define more colors in the palette (last value is not added to the gradient)

hapklaar, 2022-03-01:
@julianschill that explains it well, thank you.

For me this does what I like, which is all white lights when the printer is turned on to illuminate the bed. Then when the heater is activated start with blue and transition to red and end in white, then stop (and let the static layer take over):

```
[led_effect tool_effects]
autostart: true
frame_rate: 24
heater: extruder
leds:
    neopixel:leds
layers:
    heater  0  1 top    (0,0,1),(1,0,0),(1,1,1)
    static  0  0 add    (1,1,1)
```

closed 2022-03-01 by hapklaar