julianschill/klipper-led_effect #44: Automated LED updates in upstream klipper?

thisiscam, 2022-04-05:
Hi,

I wonder if the devs could comment on how this plugin differs from (or interact with) this new PR to Klipper: https://github.com/Klipper3d/klipper/pull/5374? Is there any thing we should be aware of when using one or the other?

thisiscam, 2022-04-05:
My main concern/question, is on this comment in that PR:

> A notable advantage of this system over custom macros using the SET_LED command is that it can update the LEDs even when the g-code command queue is blocked.

julianschill, 2022-04-05:
In Klipper you now can use display templates to create some functions to change the colors of the LEDs. I tried it out and had difficulties doing smooth animations. Also programming it is not very straight forward. The templates are only evaluated twice a second, so framerate is low.

LED-Effect has built-in effects, that you can mix and stack, while for the templates you have to program everything with Jinja2.

You can use both next to each other and at the same time, (not on the same LED at the same time, though). I also refactored the led-effect code to use the new abstraction layer the PR introduced. So it should be quite compatible.

About your concern/question: The LED effect also updates the LEDs while the queue is blocked. This was a limitation of the g-code macro approach.

closed 2022-04-12 by julianschill