julianschill/klipper-led_effect #45: Problem with colors after update

panik988, 2022-04-05:
Hello,
before the last update all the configured colors worked correctly.
After the last update instead of having the red led I have the green one.
The neopixel configuration is correct, the problem seems to be the plugin.

Thank you

julianschill, 2022-04-05:
Please specify the klipper version and the klipper-led_effect version you are using. Which color_order do your LEDs have?

panik988, 2022-04-05:
[image]

[neopixel neopixel]
pin: PE6
chain_count: 96
color_order: GRB
initial_RED: 0
initial_GREEN: 0
initial_BLUE: 0

panik988, 2022-04-05:
And this is my cfg for the led effects

[led_effect.cfg.txt](https://github.com/julianschill/klipper-led_effect/files/8416239/led_effect.cfg.txt)

julianschill, 2022-04-05:
This combination should work. Can you verify the color_order is correct without LED Effect by running:

`SET_LED LED=neopixel RED=1.0` 

Also try restarting everything.

panik988, 2022-04-05:
I tried the command SET_LED LED = neopixel RED = 1.0 and the led strip turns red.

Also restarting the PI the led strip turns red with the command so the sequence is correct.

I also tried with GREEN = 1.0 and BLUE = 1.0, all colors match

julianschill, 2022-04-05:
That's really weird. It works perfectly fine here. The new version also uses the klipper functions for the color mapping. So the only reason I can think of is that you are somehow running an old version despite the update manager saying otherwise. Can you confirm, that the led_effect.py in klipper/klippy/extras points to the correct file in the led_effect repository? 

Run `ls -l ~/klipper/klippy/extras/led_effect.py`
The output should look  like this:
`lrwxrwxrwx 1 pi pi 45 Nov 13 18:18 klipper/klippy/extras/led_effect.py -> /home/pi/klipper-led_effect/src/led_effect.py`

Also provide a klippy.log, maybe I can find something-

panik988, 2022-04-05:
pi@raspberrypi:~ $ ls -l ~/klipper/klippy/extras/led_effect.py
lrwxrwxrwx 1 pi pi 45 Feb  7 14:32 /home/pi/klipper/klippy/extras/led_effect.py -> /home/pi/klipper-led_effect/src/led_effect.py
pi@raspberrypi:~ $

Thi is the result, the date is different

julianschill, 2022-04-05:
Please run 
`cd klipper-led_effect
git log -n 1
`

panik988, 2022-04-05:
`pi@raspberrypi:~ $  cd klipper-led_effect
pi@raspberrypi:~/klipper-led_effect $  git log -n 1
commit fc11bd2a74f6e19ad5667159dbaf672b25d58e3e (HEAD -> master, origin/master, origin/HEAD)
Author: Julian Schill <j.schill@web.de>
Date:   Sat Apr 2 23:08:38 2022 +0200

    Fix effects for new klipper implementation (#43)

    * Adapted to new klipper structure and cleaned up
    * Using new LEDHelper. Fixes dotstars.
    * refactoring and simplification of getColorData
pi@raspberrypi:~/klipper-led_effect $
`

julianschill, 2022-04-05:
That looks right. 
I tried your configuration and the welcome4 effect breathes purple and red on my GRB LEDs, as it should. So I can't reproduce your problem, sorry.

panik988, 2022-04-05:
[klippy.log](https://github.com/julianschill/klipper-led_effect/files/8418437/klippy.log)

julianschill, 2022-04-05:
Everything looks good. Maybe update the firmware on the MCU, but I don't think that's the issue, as nothing changed on this side.

panik988, 2022-04-05:
OK
I'll try with a firmware update and let you know.
Thank you!

CodeMonkeyX, 2022-04-06:
Just an additional data point. I am in the process of building a Trident. I had the LED effects working on a strip, and the SB LED's. Then I updated Klipper and it stopped working. I can not debug right now because I disabled it as I am trying to finish the build. By stopped working I mean the breathing effect I had setup only lit up the first 6 LED and did not work on any other LED's in the chain after that. And the breathing was the wrong color (rainbow looking colors instead of blue).

Maybe related, I have GRBW LED's and I had a toggle macro I made that checked the neo_pixel object in the printer to see the value of W. After the update it said W was not a valid attribute.  I took a quick look in the changelog to see if anything got changed regarding LED's in Klipper and did not see anything.

panik988, 2022-04-06:
Yesterday I made a klipper update and when the firmware was restarted the colors of the led strip returned to normal.
At this point I think Klipper has been debugged which in the previous update gave problems to the plugin

CodeMonkeyX, 2022-04-06:
Thanks you are correct. I just enabled the effects again after updating and everything seems normal again. I guess something got messed up on one of the Klipper updates.

julianschill, 2022-04-06:
Glad to hear it works now.

closed 2022-04-06 by julianschill