julianschill/klipper-led_effect #46: Printer does not start up

rocketscientist89, 2022-04-06:
Hello!

I installed everything as per to the instructions but when I define the effects, on the config file, printer does not start up.

[neopixel chamber_neopixel]
pin = PB6
chain_count = 69

[led_effect panel_idle]
autostart = true
frame_rate = 24
leds = 
	neopixel:chamber_neopixel
layers = 
	breathing 10 1 bottom (0.5,0.5,1.0)

I get this message: 

Unhandled exception during run
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

Just to clarify, my LEDs work fine, and I don't believe that's the issue.

julianschill, 2022-04-06:
Please upload your klippy.log.
Also did you indent the lines after `leds` and `layers`?

rocketscientist89, 2022-04-06:
Yes, it was properly indented and with ":" not "=". Copy/Paste from the repository and then changing the name of the neopixels. The lines that I pasted above, come from the log file and not from the config. I should have stated that before.

[klippy.log](https://github.com/julianschill/klipper-led_effect/files/8430579/klippy.log)

I attached the log for you.

julianschill, 2022-04-06:
Your version of Klipper is too old. Please update to the newest Version.

rocketscientist89, 2022-04-07:
I updated my version of Klipper and now everything's working as it should.
Thanks you!

closed 2022-04-07 by rocketscientist89