julianschill/klipper-led_effect #48: Dim lights when z-axis is closer to 0

thisiscam, 2022-04-12:
Hi,

I noticed that when my z-axis is closer to the print bed, the LEDs on the extruder reflects too much light causing my streaming camera to lost all the details.
Is there a way to dim LEDs based on position of a stepper?

julianschill, 2022-04-12:
At the moment probably not. You could try out multiplying with a [Stepper ](https://github.com/julianschill/klipper-led_effect/blob/master/docs/LED_Effect.md#stepper) Effect. But I don't think you can achieve the desired behaviour with that. I will keep this issue open and see if I can come up with something in the future.

julianschill, 2022-11-15:
I added the stepper color effect, which you can use to multiply your values with a white value depending on the position.

You can use the effect rate and cutoff parameters to modify the point wher it starts dimming and the rate of dimming. Maybe experiment in the simulator.

Example:
```
steppercolor 1.0 0.0 multiply (1.00,1.00,1.00) 
gradient     1.0 1.0 top (0.0,0.0,0.0),(1.0,0.0,0.0),(0.0,1.0,0.0),(0.0,0.0,1.0) 
```

closed 2022-11-15 by julianschill