julianschill/klipper-led_effect #50: LED effects seem to cause issue with recent update

srothwel, 2022-04-18:
Love the LED effects, but I have just updated Klipper, Moonraker and the Raspberry Pi OS via the links in Fluid and now I get an error _'internal error during ready callback: PrinterNeoPixel instance has no attribute 'chain_count'_  If I comment out the effects in the config file then everything back to normal but no LED Effects.  Declaring Neopixels in the config is still okay just not the effects.

julianschill, 2022-04-18:
There were some changes in the LED code of klipper today. So LED effects might need some fix.

julianschill, 2022-04-18:
Just to be sure: Have you defined chain_count in the neopixel configuration?

meganssmith, 2022-04-19:
I got the same issue with mainsail today as well.  and yes all my neopixles had the chain_count set on them

w35t13, 2022-04-19:
wish i saw this before i hit update haha

julianschill, 2022-04-19:
As a workaround until I fix this, you can add the indices to the LED definition like this: `neopixel:my_leds (1-8)`
So e.g.:
```
[led_effect panel_idle]
autostart:              true
frame_rate:             24
leds:
    neopixel:panel_ring (1-8)
layers:
    breathing  10 1 top (.5,.5,1)
```

srothwel, 2022-04-19:
Great, thanks for the workaround tip.  Look forward to some great exciting stuff.

⁣Sent from Blue ​

***@***.***>

meganssmith, 2022-04-19:
thank you, that fixed my issues

seb449, 2022-04-19:
Parfait cela contourne bien le pb, merci

devolf92, 2022-04-23:
thanks! fix works

 ( start led # - end led #)

closed 2022-04-24 by julianschill

srothwel, 2022-10-11:
Yes, chain_count is defined.  All the effects worked perfectly before I updated klipper, has produced the same error on all three printers!

⁣Sent from Blue ​

***@***.***>

julianschill, 2022-10-11:
This should be fixed already. Please update LED Effect to the newest version. If you still have problems, please provide the klippy.log.

LazaroFilm, 2025-03-30:
Hi, I'm facing this error. adding (1-18) sends this error
```
Internal error during ready callback: invalid literal for int() with base 10: 'disco_leds(1'
```

PR #53 (2022-04-24): Fix for newest Klipper update. Fix shutdown handling.
Fixes #50
commit 5623a7d3 (2022-04-24): Fix for newest Klipper update. Fix shutdown handling. (#53)

Fixes #50
PR #54 (2022-04-24): Fix for newest Klipper update. Fix shutdown handling. (#53)
Fixes #50
commit ba509d1d (2022-04-24): Fix for newest Klipper update. Fix shutdown handling. (#53) (#54)

Fixes #50