julianschill/klipper-led_effect #51: HEATER effect only produces a static colour

NameOfTheDragon, 2022-04-19:
In principle, this effect is great but it doesn't seem to work. I have it configured like this

[...]

[neopixel chamber]
pin: P1.24
chain_count: 144
initial_RED: 0.031
initial_GREEN: 0.031
initial_BLUE: 0.031
color_order: GRB

[...]

[led_effect preheat]
leds:
    neopixel:chamber (106-140)
autostart:                          true
frame_rate:                         24
heater:                             heater_bed
layers:
    heater  45 1 top  (0,1,0),(1,0,0)

I expected this to start up green, then transition to red as the bed heats up, but it just stays a solid green until it finally goes out when the bed reaches temperature.

Maybe I'm doing it wrong?

julianschill, 2022-04-19:
The last color of the palette is used for "temperature reached". So easiest fix would be to define the second color twice.

julianschill, 2022-04-24:
Try this: 
```
[led_effect preheat]
leds:                               
    neopixel:chamber (106-140)
autostart:                          true
frame_rate:                         24
heater:                             heater_bed
layers:
    heater 45 0 top    (0,1,0),(1,0,0),(1,1,1)
```

julianschill, 2022-09-21:
@NameOfTheDragon Did you get it to work? Can this be closed?

FloxRoxX, 2022-10-09:
@julianschill 
Iam also in the state of trying out the heater effect. I also get a Static color only.
This is my LED.cfg
```
[neopixel neo]
pin: P1.24
chain_count: 20
color_order: GRB
initial_RED: 0.0
initial_GREEN: 0.4
initial_BLUE: 0.0
#initial_WHITE: 1.0

[led_effect panel_idle]
autostart:              true
frame_rate:             24
leds:
    neopixel:neo
layers:
#   layer_name  Effect Rate   Cutoff   Blending   Color
    breathing   10            1        top        (.5,.5,1)

[led_effect heater]
autostart:              true
frame_rate:             24
leds:
    neopixel:neo
heater:                 heater_bed
layers:
#   layer_name  Effect Rate   Cutoff   Blending   Color
    heater      20            0        add        (0,0,1),(1,0,0),(1,1,0)
```
Steps to reproduce:
1. Restart klipper
2. Command: STOP_LED_EFFECTS
3. Command: SET_LED_EFFECT EFFECT=heater
4. First color (0,0,1) is now applied
5. Starting to heat the heatbed to 40C
6. Color goes instantly to 1,1,0

What can i do about this? Are iam missing something ?

julianschill, 2022-10-09:
@FloxRoxX Is your bed maybe already hot? It switches to the last color, when the temperature is in the range of 2°C of the target temperature.

FloxRoxX, 2022-10-09:
@julianschill No, i started with a room temp bed and tried to raise it from 20C to about 40C

julianschill, 2022-10-09:
I cannot reproduce this. When I try your config it starts blue, turns purple, becomes red while heating up and as soon as it reaches 38°C it turns green.

closed 2023-01-28 by julianschill