julianschill/klipper-led_effect #52: "AttributeError: PrinterNeoPixel instance has no attribute 'led_helper'" When effect is in config

ryryog25, 2022-04-21:
Hi, I'm not terribly familiar with this stuff, but after installing the plugin via the provided install script and adding an effect to my printer config, I get:

```
Unhandled exception during run
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 201, in run
    self.reactor.run()
  File "/home/pi/klipper/klippy/reactor.py", line 269, in run
    g_next.switch()
  File "/home/pi/klipper/klippy/reactor.py", line 310, in _dispatch_loop
    timeout = self._check_timers(eventtime, busy)
  File "/home/pi/klipper/klippy/reactor.py", line 156, in _check_timers
    t.waketime = waketime = t.callback(eventtime)
  File "/home/pi/klipper/klippy/extras/led_effect.py", line 192, in _getFrames
    chain.led_helper.led_state[index] = (0.0, 0.0, 0.0, 0.0)
AttributeError: PrinterNeoPixel instance has no attribute 'led_helper'
Transition to shutdown state: Unhandled exception during run
```

I've got this in my config:

```
#############
# Neopixels #
#############

[neopixel rail_light]
pin: P1.24
chain_count: 25
color_order: GRB
initial_RED: .5
initial_GREEN: 0
initial_BLUE: .5

[led_effect idle]
autostart:              true
frame_rate:             24
leds:
    neopixel:rail_light (1-25)
layers:
    breathing  10 1 top (.5,.5,1)
```

I'm not quite sure if this is a bug or user error on my part, and I'm not sure what to do in order to troubleshoot this.

Thank you in advance for any help.

julianschill, 2022-04-21:
See #50.

ryryog25, 2022-04-21:
I'm using that fix, this throws a different error and the fix doesn't change anything.

ryryog25, 2022-04-21:
Unless I misunderstand what you mean in that thread.

julianschill, 2022-04-21:
Sorry I misread. Your issue is probably that your Klipper version is too old.

ryryog25, 2022-04-21:
It would appear I'm on version 0.9, so I'll update that!

ryryog25, 2022-04-23:
After updating (and fighting my mcu to update as well since that was needed for the update with klipper) the effects run without issue. I appreciate the help, and your work on this project!

lnorton89, 2023-03-01:
This issue happens for me with latest Klipper / klipper-led_effect using Octoklipper.

julianschill, 2023-03-01:
Please check your klipper version. It can be found in mainsail in the machine tab or in the klippy.log.

lnorton89, 2023-03-01:
> Please check your klipper version. It can be found in mainsail in the machine tab or in the klippy.log.

I swear I was on the latest but I was not. Thoughts on implementing a check in the install script?

closed 2024-02-21 by julianschill