julianschill/klipper-led_effect #55: After new Klipper update Printer wont boot with certain effect enabled. Zero error message just cant connect to mcu

Jo3yDaPh1sH, 2022-04-24:
When I have a Led Effect defined as Comet, Bed Temp, Progress or twinkle, the newest version of klipper v0.10.0-384-g5517a856 fails to start with only a cannot connect to mcu error. Once those lines are removed it boots fine.  So far static and linear fade have worked.

julianschill, 2022-04-24:
Could you please provide your config and a klippy.log?

julianschill, 2022-04-24:
I also just pushed some fixes for the newest Klipper version. So make sure you try the newes LED Effect version.

Jo3yDaPh1sH, 2022-04-24:
I will try the update after my print finishes. I cut the log down as it was too large for pastebin. Let me know if you need any other info.

Log
https://paste-bin.xyz/53221
Config
https://paste-bin.xyz/53222

julianschill, 2022-04-24:
`PrinterNeoPixel instance has no attribute 'chain_count' `
This is related to #50 and should be fixed by #54

Jo3yDaPh1sH, 2022-04-24:
Thank you. I love your work.

closed 2022-05-15 by julianschill