julianschill/klipper-led_effect #56: Repeated "Neopixel update did not succeed" error until klipper crashes

drb1981, 2022-05-15:
With any LED effects enabled klipper continuously gives "Neopixel update did not succeed" error until klipper overloads and crashes.  Fresh install of klipper and LED control module installed did not resolve issue.
[klippy (5).log](https://github.com/julianschill/klipper-led_effect/files/8694838/klippy.5.log)
.

julianschill, 2022-05-15:
You have a lot of LEDs. Could be a bit too much, especially with only one MCU, which also has to control the motors. I would recommend using a separate MCU for the LEDs. You could also try reducing the frame rate or amount of LEDs active at the same time.

drb1981, 2022-05-15:
@julianschill Understood, but this setup has worked fine for the past ~6 months until the most recent LED Module update?

julianschill, 2022-05-15:
Ok, that's a different story then. But I am not sure if I can do anything about it as it might be due to changes to klipper. I will look into it.

closed 2022-05-15 by julianschill

drb1981, 2022-05-15:
@julianschill I played with number of LEDs, issue appears to only happen when greater than 21 LEDs selected in chain and the greater the amount of LEDs included the more frequent the error exists.  Not sure if that is helpful in anyway, but it's what I found.     Also, MCU load is only 0.28 while this is happening according to Mainsail UI.

drb1981, 2022-05-28:
@julianschill I take it this issue isn't going to be resolved?  Loved the module until this started :(

HolyFrank, 2025-05-09:
Do you find a fix? I have the same issue.

AcrimoniousMirth, 2026-01-14:
I've been getting this for ages now with 3 NeoPixels. It seems to have worsened again after this latest Klipper update (v0.13.0-464-g48f0b3ca).