julianschill/klipper-led_effect #58: Add code for endstop state and  RPi host and MUC temperature sensors?

Polar-Ted, 2022-05-16:
I would like to have a temp status light for MCU and Pi Host CPU temps.
An endstop state would also  be useful. 

This is the line in printer.cfg that plots the temp in the on-screen graph if that helps. 

[temperature_sensor raspberry_pi]
sensor_type: temperature_host
min_temp: 10
max_temp: 100

[temperature_sensor pico]
sensor_type: temperature_mcu
sensor_mcu: pico
min_temp: 10
max_temp: 100

julianschill, 2022-05-16:
Temperature sensors will be implemented by #35
For Endstops I will have to see what is possible.

Polar-Ted, 2022-05-16:
> Temperature sensors will be implemented by #35 For Endstops I will have to see what is possible.

Maybe I'm writing the effect script incorrectly

When I define  **heater: "temperature_sensor pico"**   the  firmware fails to start with this error

_Internal error during ready callback: Unknown heater 'temperature_sensor pico'
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown_

julianschill, 2022-05-16:
No it is not yet implemented. Your issue is just a duplicate of #35

burntcoinpurse, 2022-07-13:
I got #35 figured out. 👍

luisfosoares, 2022-08-10:
Any progress on this? ETA? I would appreciate it so much thanks

julianschill, 2022-09-21:
No ETA. It is done, when it's done.

julianschill, 2023-02-04:
Endstops can now trigger the new homing effect.

closed 2023-02-04 by julianschill