julianschill/klipper-led_effect #60: Effet K2000

seb449, 2022-05-28:
Hi,

is it possible to make an effect like k2000, comet in one direction then in the other ?
Thanks

seb449, 2022-05-28:
[led_effect Comet_Rouge]
#frame_rate:             24
leds:
    neopixel:my_neopixel (1-51)
layers:
    comet 1 2 add (1.0, 0.0, 0.0),(1.0, 0.0, 0.0)
    comet -1 2 add (1.0, 0.0, 0.0),(1.0, 0.0, 0.0)

seb449, 2022-05-28:
but two comet !

Dito-82, 2022-06-20:
I experimented a bit with this to try and make a reflection of the comets work.
Seems like the comet animation is always 26 pixel long, regardless of setting for tail length.
The mirrored comet starts as soon as the first one is 26 pixel into its animation.
To make this work, the defined strip section has to be 26 long.

This works fine for me:

[led_effect comet_rouge]
leds:              
  neopixel:neo (1-26)
layers:
   comet 0.5 1 add (1.0, 0.0, 0.0),(1.0, 0.0, 0.0)
   comet -0.5 1 add (1.0, 0.0, 0.0),(1.0, 0.0, 0.0)

You could even try and spread the animation again to nearly your original size with:

[led_effect comet_rouge_spread1]
leds:                  
   neopixel:neo (1,3,5,7,9,11,13,15,17,19,21,23,25,27,29,31,33,35,37,39,41,43,45,47,49,51)
layers:
   comet 0.5 1 add (1.0, 0.0, 0.0),(1.0, 0.0, 0.0)
   comet -0.5 1 add (1.0, 0.0, 0.0),(1.0, 0.0, 0.0)

and even fill in the remaining leds with a second animation and calling both with a macro:

[led_effect comet_rouge_spread2]
leds:                  
   neopixel:neo (2,4,6,8,10,12,14,16,18,20,22,24,26,28,30,32,34,36,38,40,42,44,46,48,50,52)
layers:
   comet 0.5 1 add (1.0, 0.0, 0.0),(1.0, 0.0, 0.0)
   comet -0.5 1 add (1.0, 0.0, 0.0),(1.0, 0.0, 0.0)

[gcode_macro kitt]
gcode:
   SET_LED_EFFECT EFFECT=comet_rouge_spread1
   SET_LED_EFFECT EFFECT=comet_rouge_spread2

But be warned, this is very heavy on the mcu. 
My mcu went up to 50% load just by running the macro

bybarbaros, 2024-02-19:
Hello, let me revive the topic. :)
comet 0.5 1 add (1.0, 0.0, 0.0),(1.0, 0.0, 0.0)
comet -0.5 1 add (1.0, 0.0, 0.0),(1.0, 0.0, 0.0)

  I applied these values to the macro. in conclusion. Both the left and the right began to work together at the same time. **from left to right. **did not move from right to left.

bybarbaros, 2024-02-19:
for your help. Thank you very much in advance .

burntcoinpurse, 2024-02-19:
Try making 2 separate comet macros, that worked for me.

bybarbaros, 2024-02-19:
[led_effect sb_logo_kitt]
autostart:               false
frame_rate:              24
leds:
    neopixel:sb_leds (23-49)
layers: 
 comet 1 2 add (1.0, 0.0, 0.0),(1.0, 0.0, 0.0)
 comet -1 2 add (1.0, 0.0, 0.0),(1.0, 0.0, 0.0)
************
[gcode_macro LED_KITT]
gcode:
    STOP_LED_EFFECTS
    SET_LED_EFFECT EFFECT=sb_logo_kitt  # FADETIME=3.0 REPLACE=5
    set_nozzle_leds_on

https://github.com/julianschill/klipper-led_effect/assets/118777939/b5a3c921-f45f-4764-a134-14d88582fdec

bybarbaros, 2024-02-19:
> Try making 2 separate comet macros, that worked for me.

Believe me, I would like to do it too.. but... the "but" here is important. My software knowledge level. Just copy and paste of course. My most important supporter is Google Translate.

bybarbaros, 2024-02-19:
I tried to change the settings.. I tried different variations but I was not successful.

julianschill, 2024-02-21:
Thanks to Viktor von Due (on Discord) we got this example which seems to work quite nicely:
`gradient -0.50 0.50 add (1.00,0.00,0.00),(0.00,0.00,0.00),(0.00,0.00,0.00),(0.00,0.00,0.00) 
gradient 0.50 0.60 top (1.00,0.00,0.00),(0.00,0.00,0.00),(0.00,0.00,0.00),(0.00,0.00,0.00) `

bybarbaros, 2024-02-24:
It's very good... it works perfectly, thank you very much indeed. It was the legendary TV series of my childhood. Even my phone's ringtone is the background music of the series. Adding such a configuration to the system really made me very happy. I would like to thank everyone who contributed.

DVSVIDEO, 2025-01-03:
chase 1.60 2.00 add (1.00,1.00,1.00),(1.00,0.00,0.00) 
 chase -1.60 2.00 add (1.00,1.00,1.00),(1.00,0.00,0.00)

bybarbaros, 2025-01-05:
&gt; My bed mesh effect: chase 1.60 2.00 add (1.00,1.00,1.00),(1.00,0.00,0.00) chase -1.60 2.00 add (1.00,1.00,1.00),(1.00,0.00,0.00) https://youtu.be/ihvW4aQfM9A?si=k4ofskDrpJgTU2mJ

I will try. thanks