julianschill/klipper-led_effect #64: Error config

dagov42, 2022-06-24:
I put into config this code from docs:
```
[led_effect critical_error]
leds:
    neopixel:tool_lights
    neopixel:bed_lights
layers:
    strobe         1  1.5   add        (1.0,  1.0, 1.0)
    breathing      2  0     difference (0.95, 0.0, 0.0)
    static         1  0     top        (1.0,  0.0, 0.0)
autostart:                             false
frame_rate:                            24
run_on_error:                          true
```

And I have error:
```
Internal error during ready callback: 'float' object cannot be interpreted as an integer
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
```
In log:
```
Traceback (most recent call last):
  File "/home/klipper/klipper/klippy/klippy.py", line 205, in _connect
    cb()
  File "/home/klipper/klipper/klippy/extras/led_effect.py", line 382, in _handle_ready
    self.layers.insert(0, layer(handler       = self,
  File "/home/klipper/klipper/klippy/extras/led_effect.py", line 659, in __init__
    for c in range(0, len(self.paletteColors)):
TypeError: 'float' object cannot be interpreted as an integer
Transition to shutdown state: Internal error during ready callback: 'float' object cannot be interpreted as an integer
```

julianschill, 2022-06-25:
Are you using Python 3?

dagov42, 2022-06-25:
Yes, 3.9

julianschill, 2022-06-25:
Ok, I never tested it with Python 3. But I think I found the issue. But there may be more problems.

dagov42, 2022-06-25:
May be you can help me to fix this problem locally? I've tried to use float parameters and integer. It's doesn't work.

julianschill, 2022-06-25:
Should be fixed now. Try updating and let me know if there are more bugs.

dagov42, 2022-06-25:
```
Internal error during ready callback: unsupported operand type(s) for *: 'NoneType' and 'int'
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
```

```
Unhandled exception during run
Traceback (most recent call last):
  File "/home/klipper/klipper/klippy/klippy.py", line 217, in run
    self.reactor.run()
  File "/home/klipper/klipper/klippy/reactor.py", line 269, in run
    g_next.switch()
  File "/home/klipper/klipper/klippy/reactor.py", line 310, in _dispatch_loop
    timeout = self._check_timers(eventtime, busy)
  File "/home/klipper/klipper/klippy/reactor.py", line 156, in _check_timers
    t.waketime = waketime = t.callback(eventtime)
  File "/home/klipper/klipper/klippy/extras/led_effect.py", line 226, in _getFrames
    next_eventtime=min(self.effects, key=lambda x: x.nextEventTime)\
ValueError: min() arg is an empty sequence
```

julianschill, 2022-06-25:
I commited some fixes to make it work with Python 3. I hope this solves it for you now.

dagov42, 2022-06-26:
```
Unhandled exception during run
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
```

```
Starting heater checks for extruder
Starting heater checks for heater_bed
Stats 122.0: gcodein=0 mcu: mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000 bytes_write=4023 bytes_read=5494 bytes_retransmit=9 bytes_invalid=0 send_seq=250 receive_seq=238 retransmit_seq=2 srtt=0.002 rttvar=0.003 rto=0.025 ready_bytes=1095 stalled_bytes=70 freq=63992991  heater_bed: target=0 temp=0.0 pwm=0.000 sysload=1.62 cputime=9.954 memavail=746132 print_time=0.001 buffer_time=0.000 print_stall=0 extruder: target=0 temp=0.0 pwm=0.000
Unhandled exception during run
Traceback (most recent call last):
  File "/home/klipper/klipper/klippy/klippy.py", line 217, in run
    self.reactor.run()
  File "/home/klipper/klipper/klippy/reactor.py", line 269, in run
    g_next.switch()
  File "/home/klipper/klipper/klippy/reactor.py", line 310, in _dispatch_loop
    timeout = self._check_timers(eventtime, busy)
  File "/home/klipper/klipper/klippy/reactor.py", line 156, in _check_timers
    t.waketime = waketime = t.callback(eventtime)
  File "/home/klipper/klipper/klippy/extras/led_effect.py", line 134, in _pollHeater
    for heater in self.heaters.iterkeys():
AttributeError: 'dict' object has no attribute 'iterkeys'
Transition to shutdown state: Unhandled exception during run
Dumping gcode input 0 blocks
Dumping 8 requests for client 547940012768
Received 120.830755: b'{"id": 548551750992, "method": "objects/subscribe", "params": {"objects": {"webhooks": null, "print_stats": null, "heater_bed": null, "extruder": null}, "response_template": {"method": "process_status_update"}}}'
Received 120.853812: b'{"id": 548551750992, "method": "gcode/subscribe_output", "params": {"response_template": {"method": "process_gcode_response"}}}'
Received 120.857527: b'{"id": 548551751328, "method": "info", "params": {"client_info": {"program": "Moonraker", "version": "v0.7.1-589-g4bed314"}}}'
Received 121.113268: b'{"id": 548551750224, "method": "info", "params": {}}'
Received 121.367428: b'{"id": 548551750224, "method": "info", "params": {}}'
Received 121.621413: b'{"id": 548551750128, "method": "info", "params": {}}'
Received 121.735726: b'{"id": 548551750992, "method": "info", "params": {}}'
Received 121.875984: b'{"id": 548551750992, "method": "info", "params": {}}'
gcode state: absolute_coord=True absolute_extrude=True base_position=[0.0, 0.0, 0.0, 0.0] last_position=[0.0, 0.0, 0.0, 0.0] homing_position=[0.0, 0.0, 0.0, 0.0] speed_factor=0.016666666666666666 extrude_factor=1.0 speed=25.0
Reactor garbage collection: (121.569062974, 0.0, 0.0)
```

dagov42, 2022-06-28:
I solve it. I change iterkeys() to keys().

closed 2022-06-29 by dagov42