julianschill/klipper-led_effect #72: Please tell me how to install manually

Alan-0131, 2022-07-16:
Hello, I like this LED-effect very much. I use Klippy in Repetier-Server. I artificially copied the led_effect.py file to the directory required by Repetier-Server /opt/klipper/<Printer slug>/klippy/extras , only the following code can run
[neopixel sb_leds]
pin: PD3
chain_count: 13
color_order: GRBW
 
initial_RED: 0
initial_GREEN: 1
initial_BLUE: 1
initial_WHITE: 1
Can't add macros anymore below, please tell me how to install manually instead of script, thanks again.

julianschill, 2022-08-14:
Please provide your config and the klippy.log.

julianschill, 2022-08-31:
Copying led_effect.py into klippy/extras should be enough. You should then be able to add effects. See documentation for simple examples.

julianschill, 2022-09-21:
Can this be closed?

julianschill, 2022-10-23:
Closing because of inactivity. Feel free to reopen or open a new one, when further support is needed.

closed 2022-10-23 by julianschill