julianschill/klipper-led_effect #75: Effect affects all LEDs not just defined

Komet, 2022-08-04:
Hi, thanks for your work! Really a nice addition :)

I have two LED strips on different pins and an effect defined for one of them. If I start the effect it also toggles the LEDs of the other strip. This means: Setting the toolhead LEDs to red and then starting the fire effect, the toolhead LEDs will turn off. I don't think that this is the intended behaviour (?) I would assume that the other LEDs would simply be not affected.

```[neopixel top_lights]
pin: PB6
chain_count: 17
color_order: GRB

[neopixel toolhead]
pin: PB7
chain_count: 3
color_order: GRBW

[led_effect fire]
frame_rate: 24
leds:
    neopixel:top_lights
layers:
    fire 45 40 top (0.0, 0.0, 0.0),(1.0, 0.0, 0.0),(1.0, 1.0, 0.0),(1.0, 1.0, 1.0)
```

julianschill, 2022-08-14:
I suppose you set the other LEDs to red using SET_LED? I have to look into this. As a workaround, you could try defining an effect with static red and run that. Let me know if this works.

julianschill, 2022-08-14:
Also: Can you provide a log and/or your complete configuration?

julianschill, 2022-09-21:
@Komet Any news on that? I can't reproduce it on my side.

julianschill, 2022-10-23:
Closing because of inactivity. Feel free to reopen or open a new one, when further support is needed.

closed 2022-10-23 by julianschill