julianschill/klipper-led_effect #76: Calculation of stepper position (as %) is incorrect for non-zero min axis

guru-florida, 2022-08-08:
This percent calculation seems incorrect on [line 152](https://github.com/julianschill/klipper-led_effect/blob/f12dcdc21d10dafa0e8788b9869d0c4b09c0c308/src/led_effect.py#L152):
```python
 self.stepperPositions[i] = int(
                    (pos[i] / (self.kin.axes_max[i] - self.kin.axes_min[i])
                        * 100)- 1)
```

The pos[i] range would be between min < max, but the denominator part of the division is the magnitude of {min,max}...therefor `pos[i]` should be `(pos[i] - self.kin.axes_min[i])` so that the numerator becomes a value between 0 and the magnitude of {min,max}. 

So I believe the code should be:
```python
 self.stepperPositions[i] = int(
                    ((pos[i] - self.kin.axes_min[i]) / (self.kin.axes_max[i] - self.kin.axes_min[i])
                        * 100)- 1)
```

That would get you a percentage between 0 and 1...except that trailing -1 is weird...so perhaps a final value between -1 and 99.

julianschill, 2022-08-31:
I think it has to be `(pos[i] + self.kin.axes_min[i])`.

guru-florida, 2022-09-01:
If you want to have a percentage between min and max, then I would say it needs to subtract the axis_min...you also subtract the axis_min in the denominator as well.

commit f244ad11 (2022-09-21): Calculation of stepper position (as %) is incorrect for non-zero min axis
Fixes #76
PR #87 (2022-09-21): Calculation of stepper position (as %) is incorrect for non-zero min …
…axis

Fixes #76
closed 2022-09-21 by julianschill