julianschill/klipper-led_effect #77: How do I start with one effect then after a time, automatically switch to another?

Gliptopolis, 2022-08-09:
I see the FADETIME option but it seems that has to be tied to a macro? I'd like the machine to boot up with one effect then after a couple seconds switch to another one. Imagine a comet effect for a couple seconds then switching to a static color.

gdachs, 2022-08-09:
IMHO this is not an issue, because you can do that without any change to klipper-led_effect. Just use https://www.klipper3d.org/Config_Reference.html#idle_timeout

burntcoinpurse, 2022-08-10:
You can set a gcode command in octoprint to initiate on the serial connection. If that helps.

Gliptopolis, 2022-08-10:
> You can set a gcode command in octoprint to initiate on the serial connection. If that helps.

I'm using Mainsail so not sure if this carries over or not? I can't find any references to this in klipper docs

julianschill, 2022-08-14:
Use delayed gcode for that. https://www.klipper3d.org/Config_Reference.html?#delayed_gcode

Gliptopolis, 2022-08-14:
I have a delayed gcode in my printer.cfg but it doesn't do anything as far as I can tell.

julianschill, 2022-08-31:
Set autostart to true for the first effect and start the second effect with the delayed_gcode like this:

`[delayed_gcode startup_effect]
initial_duration: 3
gcode:
  SET_LED_EFFECT EFFECT=first_effect STOP=1
  SET_LED_EFFECT EFFECT=second_effect`

closed 2022-09-21 by julianschill