julianschill/klipper-led_effect #81: addressing different LEDs in a string

ArketypeDesign, 2022-09-07:
I am trying to combine different effects on a stealthburner toolhead (3 LEDs...Logo is led 1, toolhead lights are 2,3)

not sure this syntax is right and would love some guidance.

The code below only activates the logo #1. 
The toolhead leds (2,3) seem to be ignored.

[led_effect status-ready]
leds:
    neopixel:sb_leds (1)
layers:
    static         1  0      bottom (0.5,  0.0, 0.5)
autostart:                             true
frame_rate:                            24

    neopixel:sb_leds (2,3)
layers:
    breathing      8  0     add (0.4, 0.4, 0.4)
    static         1  0     bottom (0.1,  0.1, 0.1)
autostart:                             true
frame_rate:                            24

burntcoinpurse, 2022-09-08:
Try this
```
[neopixel sb_leds]
pin: "your pin"
chain_count:3
color_order:GRB
initial_GREEN: 0
initial_RED: 0
initial_BLUE: 0

[led_effect status-ready]
leds:
neopixel:sb_leds (1)
layers:
static 1 0 bottom (0.5, 0.0, 0.5)
autostart: true
frame_rate: 24

[led_effect status-ready1]
leds:
neopixel:sb_leds (2-3)
layers:
breathing 8 0 add (0.4, 0.4, 0.4)
static 1 0 bottom (0.1, 0.1, 0.1)
autostart: true
frame_rate: 24

[gcode_macro LEDS_OFF]
gcode:
   STOP_LED_EFFECTS
   SET_LED LED=sb_leds RED=0 GREEN=0 BLUE=0
```

julianschill, 2022-09-21:
@ArketypeDesign Can this be closed?

julianschill, 2022-10-23:
I will close this because of inactivity.

closed 2022-10-23 by julianschill