julianschill/klipper-led_effect #82: config error

dagov42, 2022-09-09:
After update klipper to v0.10.0-571-g97a5b39a led-effect dont work. 
Log is: 

> Config error
> Traceback (most recent call last):
>   File "/home/klipper/klipper/klippy/klippy.py", line 175, in _connect
>     self._read_config()
>   File "/home/klipper/klipper/klippy/klippy.py", line 145, in _read_config
>     pconfig.check_unused_options(config)
>   File "/home/klipper/klipper/klippy/configfile.py", line 295, in check_unused_options
>     raise error("Section '%s' is not a valid config section"
> configparser.Error: Section 'led_effect preheat' is not a valid config section
> webhooks client 547694102464: New connection
> webhooks client 547694102464: Client info {'program': 'Moonraker', 'version': 'v0.7.1-660-g5d856b9'}

julianschill, 2022-09-09:
Please try reinstalling led_effect.

julianschill, 2022-10-23:
@dagov42 Is it working for you now? Can this be closed?

dagov42, 2022-10-23:
Yes, you can do it. It was klipper updating bug.

closed 2022-10-23 by julianschill