julianschill/klipper-led_effect #83: What is the correct stepper format?

AdamV3D, 2022-09-12:
I've tried a number of different formats for specifying the stepper motor but it always seems to respond to Z regardless of what I try.

What is the correct way to specify the Y or X motor for example?

`[led_effect idle]
autostart: true
frame_rate: 24
leds:
    neopixel:side_light
stepper: stepper_x
layers:
   stepper     0 0 top (1.0, 0.0, 0.0)`

julianschill, 2022-09-21:
Like this:

[led_effect stepper]
leds:                               
    neopixel:side_light 
layers:            
    stepper   2  2 top (1, 1, 1),(0, 0, 1),(0, 1, 0)  
frame_rate:                           24
autostart:                             true
stepper: x

julianschill, 2022-10-06:
Looking at your video, I think this can be closed.

closed 2022-10-06 by julianschill