julianschill/klipper-led_effect #89: Create LED group module

julianschill, 2022-09-21:
A module that uses LED helper to group LEDs together. Example:
```
[led_group my_group]
Leds: neopixel:my_neo (1-3)
    neopixel:my_other_neo(8-5)
```
Can than be used like any other LED chain in SET_LED and SET_LED_EFFECT.

julianschill, 2022-09-25:
Pull request to klipper: https://github.com/Klipper3d/klipper/pull/5788

Okseklump, 2023-07-19:
Hey!
Is it possible to use these groups so that my 2 bars, and 1 Neopixel are synced? Right now its goes 1-2-3, i want them to show the same effect all at once.
Thanks for all your cool work on this!

julianschill, 2023-07-20:
Not sure what you mean. This is an open issue which is not implemented yet. Also syncing of two led chains is not directly possible. Easiest way is to just use two effects and start them at the same time.

thetic, 2023-12-14:
I _really_ like this idea. Is it dead?

julianschill, 2023-12-14:
The LED groups have not been merged to klipper. For now I won't proceed with it. The LED effect project itself is not dead, I just don't have a lot of time for it at the moment.

thetic, 2024-02-20:
This exists and accomplishes the same purpose: https://github.com/MapleLeafMakers/klipper-virtual_leds/