julianschill/klipper-led_effect #91: 'Heater' effect does not work (at least on RatOS)

Tomtim1, 2022-09-29:
Everything works for me except "heater" effect. It should make a nice gradient from cold to hot, but the only thing it does for me is one color, and the second only for the exact set temparure, when it reaches it. So like "blue" when it's between 0-69 degrees and "red" when it reaches 70.

My config for this one is like this:
```
[led_effect bed_effects]
leds:
    neopixel:bed_lights
autostart:                          false
frame_rate:                         30
heater:                             heater_bed
layers:
    heater  1 0 add    (0,0,1),(1,0,0)
```

And command for it is placed in START_PRINT macro:

`
STOP_LED_EFFECTS
SET_LED_EFFECT EFFECT=bed_effects`

Tomtim1, 2022-09-29:
Yep, it works, thank you very much! By the way, just a quick question after reading documentation, is there a possibility to make LEDs "load up", as temperature goes up? Like from turned off leds, from left to right for example, one by one.

julianschill, 2022-09-29:
Not yet. But it is planned. Related issue is https://github.com/julianschill/klipper-led_effect/issues/25

closed 2022-09-29 by julianschill