julianschill/klipper-led_effect #94: maximum recursion depth exceeded while calling a Python object

jiatinglu99, 2022-10-06:
Hi,

I'm getting printer shutdown upon runtime exception, the LED effect would run for a few seconds then cause an exception in klippy and shutdown the printer.

Here is the klippy log:
`Stats 47028.1: gcodein=0  mcu: mcu_awake=0.284 mcu_task_avg=0.000155 mcu_task_stddev=0.000000 bytes_write=223096 bytes_read=39756 bytes_retransmit=1233 bytes_invalid=0 send_seq=5317 receive_seq=5309 retransmit_seq=5316 srtt=0.001 rttvar=0.001 rto=0.400 ready_bytes=0 stalled_bytes=2 freq=180002231 EBBCan: mcu_awake=0.001 mcu_task_avg=0.000013 mcu_task_stddev=0.000013 bytes_write=1362 bytes_read=6241 bytes_retransmit=48 bytes_invalid=0 send_seq=161 receive_seq=159 retransmit_seq=161 srtt=0.002 rttvar=0.000 rto=1.600 ready_bytes=0 stalled_bytes=1 freq=48000141 adj=47999512 host: mcu_awake=0.001 mcu_task_avg=0.000010 mcu_task_stddev=0.000014 bytes_write=1018 bytes_read=5039 bytes_retransmit=0 bytes_invalid=0 send_seq=144 receive_seq=144 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=49999573 adj=49999030 Octopus_Pro_446: temp=27.4 Raspberry_Pi: temp=51.6  heater_bed: target=0 temp=24.2 pwm=0.000 sysload=0.47 cputime=9.342 memavail=1383500 print_time=23.723 buffer_time=0.000 print_stall=0 extruder: target=0 temp=24.2 pwm=0.000
Stats 47029.1: gcodein=0  mcu: mcu_awake=0.284 mcu_task_avg=0.000155 mcu_task_stddev=0.000000 bytes_write=223096 bytes_read=39756 bytes_retransmit=1233 bytes_invalid=0 send_seq=5317 receive_seq=5309 retransmit_seq=5316 srtt=0.001 rttvar=0.001 rto=0.400 ready_bytes=0 stalled_bytes=3 freq=180002231 EBBCan: mcu_awake=0.001 mcu_task_avg=0.000013 mcu_task_stddev=0.000013 bytes_write=1362 bytes_read=6241 bytes_retransmit=48 bytes_invalid=0 send_seq=161 receive_seq=159 retransmit_seq=161 srtt=0.002 rttvar=0.000 rto=1.600 ready_bytes=0 stalled_bytes=2 freq=48000141 adj=47999521 host: mcu_awake=0.001 mcu_task_avg=0.000010 mcu_task_stddev=0.000014 bytes_write=1024 bytes_read=5055 bytes_retransmit=0 bytes_invalid=0 send_seq=145 receive_seq=145 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=49999593 adj=49999062 Octopus_Pro_446: temp=27.4 Raspberry_Pi: temp=52.1  heater_bed: target=0 temp=24.2 pwm=0.000 sysload=0.47 cputime=9.464 memavail=1383772 print_time=23.723 buffer_time=0.000 print_stall=0 extruder: target=0 temp=24.2 pwm=0.000
Unhandled exception during run
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 217, in run
    self.reactor.run()
  File "/home/pi/klipper/klippy/reactor.py", line 292, in run
    g_next.switch()
  File "/home/pi/klipper/klippy/reactor.py", line 340, in _dispatch_loop
    timeout = self._check_timers(eventtime, busy)
  File "/home/pi/klipper/klippy/reactor.py", line 158, in _check_timers
    t.waketime = waketime = t.callback(eventtime)
  File "/home/pi/klipper/klippy/extras/led_effect.py", line 227, in _getFrames
    chain.led_helper.update_func(chain.led_helper.led_state, None)
  File "/home/pi/klipper/klippy/extras/neopixel.py", line 107, in update_leds
    self.printer.get_reactor().register_callback(reactor_bgfunc)
  File "/home/pi/klipper/klippy/reactor.py", line 169, in register_callback
    rcb = ReactorCallback(self, callback, waketime)
  File "/home/pi/klipper/klippy/reactor.py", line 43, in __init__
    self.timer = reactor.register_timer(self.invoke, waketime)
  File "/home/pi/klipper/klippy/reactor.py", line 124, in register_timer
    timer_handler = ReactorTimer(callback, waketime)
RuntimeError: maximum recursion depth exceeded while calling a Python object
Transition to shutdown state: Unhandled exception during run
Dumping gcode input 0 blocks
Dumping 20 requests for client 3044643944`

It seems like too many recursion in led_effect.py and neopixel.py?

Here is my LED effect config just in case:
`# [neopixel printer_led]
# pin: printer_neopixel_pin
# chain_count: 72
# color_order: GRB

# [output_pin led_relay]
# pin: host:gpiochip0/gpio20
# value: 1
# shutdown_value: 0

[led_effect panel_idle]
autostart: true
frame_rate: 24
leds:
    neopixel:printer_led
layers:
    gradient 0.2 1 top (1.0, 0.0, 0.0),(0.0, 1.0, 0.0),(0.0, 0.0, 1.0)`

Moonraker.log:
`2022-10-06 09:22:33,464 [proc_stats.py:_handle_shutdown()] - Throttled Flags: 
2022-10-06 09:22:33,705 [klippy_connection.py:_on_connection_closed()] - Klippy Connection Removed`

One note is that I use a relay to turn on PSUs for LED and OctopusPro, so led_effect may be running even tho the two devices aren't powered.

Any help is appreciated.

julianschill, 2022-10-06:
I experience this problem, when trying to use too many LEDs. But 72 should actually be fine. I will look into this. On which host is running klipper? A raspberry pi?

jiatinglu99, 2022-10-06:
yes I'm running on raspberry pi 4. This actually only started happening after a recent update of everything(including led_effect, klipper, moonraker, os-packages). I had to change gradient numbers to get the same effect. Back in July, it was working perfectly.

julianschill, 2022-10-06:
I reworked the gradient effect to be smoother, but it also takes more resources. You can try reducing the framerate to compensate.

jiatinglu99, 2022-10-07:
lowering the framerate did fix the issue for now. Thanks

allenrowand, 2022-10-08:
Having this issue as well, specifically while running a comet effect during nozzle scrubbing. This is on 12 LEDs across two chains. I'll try reducing the framerate, happy to provide any info that could help. Thanks!

julianschill, 2022-10-08:
@jiatinglu99 and @allenrowand which board do you use how is it connected to your Pi? USB, UART or CANBUS?

allenrowand, 2022-10-08:
I have one chain of two LEDs on an Octopus connected via USB, another chain of ten on a CAN toolhead board. I think part of my problem is that I am calling two effects too close together and the system is trying to run them simultaneously- reducing the framerate on the effect that I thought was causing the exception didn't help. I have a couple prints to finish today but I will adjust my print start script, see if that fixes the problem and report back.

Murilolana, 2023-02-27:
I'm experiencing the same thing. The breathe effect runs smoothly while not printing, but it crashes as soon as the print job starts. I've reduced the fps to 12 and it worked, but badly. The effect would skip some steps or freeze while homing and moving the print head. I'm using a SKR Pro v1.2 with Rpi3B+. Any news on this issue?

intensivecarebear80, 2023-06-13:
I might be a little late to the party but i found a simple solution to this problem as i was facing it myself. I started by removing all my Neopixel macros as they were obviously causing the issue. Then I re-added each one of my macros until i found the culprit causing the crash, then instead of messing with the framerate or anything, i simply added a G4 P500 command between the macros that were trying to be run at the same time and hey presto, problem fixed 😄