julianschill/klipper-led_effect #95: Internal error during ready callback:

BelgianGuyWithALaser, 2022-10-07:
Hello,

After using this for a day or so I started tweaking some more effects for my usecase.
While doing so I pressed the save and restart button while an effect was still active.
Resulting in 
`Internal error during ready callback:
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown`

If I go into my loggfile I can find the following error mentioned : 
`Unhandled exception during ready callback
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 205, in _connect
    cb()
  File "/home/pi/klipper/klippy/extras/led_effect.py", line 397, in _handle_ready
    
  File "/home/pi/klipper/klippy/extras/led_effect.py", line 740, in __init__`

removing the led_effects plugin solves this and I can re-use my printer. doing a fresh install of the led_effect plugin now causes the same behaviour of the above message. (I believe that it has written something in the MCU memory?)

any way to solve this without me needing to reflash both my MCU's?
[klippy (2).log](https://github.com/julianschill/klipper-led_effect/files/9733956/klippy.2.log)

julianschill, 2022-10-07:
From your log it looks like there is a bug in the chase effect. So as a workaround for now don't use the chase effect until I had time to look into it. Nothing is saved in the MCU memory, so reflashing won't be necessary.
MemoryError means, that you ran out of memory. Almost all effects are precalculated at startup and put into memory, so it could also be, that you just have defined too many effects.

BelgianGuyWithALaser, 2022-10-07:
The strange part is, the chase was working right before. That was in the effect that was active while I pressed the save and restart firmware button.
I just tried it with commenting out all the chase effects I had in there and it booted up, but my electronic fans instantly ramp up to 100% instead of the specified % in my config file.
Also trying any other kind of effect gives me an error in the command logg : !! Unhandled exception during run (even STOP_LED_EFFECTS)

so something else is going on aswell

julianschill, 2022-10-07:
It doesn't matter if there is an effect active, when restarting klipper. I think that was just a coincidence. Can you check the memory consumption of your Pi? It is in mainsail in the Machine tab.

BelgianGuyWithALaser, 2022-10-07:
but wouldn't a fresh install of the plugin resolve the issue of it's within the python file(?) itself

[image]

julianschill, 2022-10-07:
> but wouldn't a fresh install of the plugin resolve the issue of it's within the python file(?) itself

The python file is not corrupted. So reinstallation won't help. Something on your system is consuming too much memory. Could be the led_effect plugin, but could also be something else. My theory currently is, that one of your effects (or all of them combined) need too much memory. And as the error is always happening during initialiation of the chase effect, I suspect that that is the culprit. Can you provide me your complete config, so that I can try your configuration on my machine?

BelgianGuyWithALaser, 2022-10-07:
my whole printer config? or just the neopixels + effects config? I just find it strange that it was working before, but after tweaking some colours to my liking suddenly it went like this

julianschill, 2022-10-07:
Neopixels and effects should be enough.

BelgianGuyWithALaser, 2022-10-07:
Here you go.
Thanks for looking into it by the way.
[VoronNeopixels.zip](https://github.com/julianschill/klipper-led_effect/files/9734643/VoronNeopixels.zip)

julianschill, 2022-10-07:
Which version of the plugin are you using?

BelgianGuyWithALaser, 2022-10-07:
according to my software update page in fluidd latest one

julianschill, 2022-10-07:
So I just verified, that the latest update, that I pushed yesterday introduced a bug in the chase effect, that makes it use a lot of memory. I will investigate further and hopefully fix it later today.

julianschill, 2022-10-07:
It seems to happen, when there is only one color defined in the palette, like in your case_homing effect.

julianschill, 2022-10-07:
And the bug was also present in older versions. So add a second color for the "tail" of your chase to the effect and your memory problem should be gone. I will add some safety mechanism to prevent the error.

BelgianGuyWithALaser, 2022-10-07:
Just tried it, and it's working.
Shame that my eye didn't capture that. thanks for the help!

julianschill, 2022-10-07:
No shame. The effect should handle that or at least give a meaningful error message. Thanks for revealing the issue.

commit be20a730 (2022-11-06): Fix chase effect
Fixes #95
closed 2022-11-15 by julianschill