julianschill/klipper-led_effect #100: Add heater, stepper and progress to simulator

julianschill, 2022-10-23:
Add some control for heater, stepper and progress to also simulate effects that rely on those.

GadgetAngel, 2022-11-05:
Since all Voron printers have known length on x, y, and z.  I can create a palette of colors on for x and one for y and one for z that have 300 color values.  My printer is 300mm build.

What I want is the ability to use the position of the stepper motors to access the corresponding color value.

So say I setup red to be for my x-axis and green for my y-axis and my blue for my z-axis.   I want the current x,y,z position to be used so that the color produced (solid) for my strip will be {red}= xpos mod 255/300, {green} = ypos mod 255/300, and {blue} = zpos mod 255/300. This would mean the rgb value = {red%,green%,blue%}

or you can just use the xpos, ypos, zpos to access the color palette to get the r,g,b to apply for the whole led_strip. 

I have 16 leds on my stealthburner and I would like to see them change colors based on the position of the toolhead.

julianschill, 2022-11-05:
I think that needs a new effect, that changes the color instead of the position. I can see what I can do.

GadgetAngel, 2022-11-05:
> 

you could call it stepperxyz.  Basically the color would be the same on all axis you would just need to use the stepper position as an index into one array where the index would be ((stepper_position) mod 255)/(max pos))

julianschill, 2022-11-06:
@GadgetAngel I will track your request in this issue: #108 

This issue is about the simulator.

PR #107 (2022-11-06): Add heater, stepper and progress to simulator
Fixes #100
commit 0ae6ac92 (2022-11-06): Add heater, stepper and progress to simulator (#107)

Fixes #100
closed 2022-11-15 by julianschill