julianschill/klipper-led_effect #106: Example for daisy chained LEDs

andreklug, 2022-11-03:
Hi @julianschill 

could you please provide an example on how the configuration would look like for daisy chained LEDs on a single PIN? For example PB0 on a Octopus 1.1 with four segments of WS2812b. 

Thank you for your help!

julianschill, 2022-11-03:
That is just one chain and effect on different segments:

```
[neopixel awesome_led_name]
pin: PB0
chain_count: 32
initial_RED: 0.0
initial_GREEN: 0.0
initial_BLUE: 0.0
initial_WHITE: 0.0
color_order: GRB

[led_effect first_segment_rgb]
leds:
    neopixel:awesome_led_name (1-8)
layers: 
    gradient       1 1  top  (1.0, 0.0, 0.0),(0.0, 1.0, 0.0),(0.0, 0.0, 1.0)
autostart:                             false
frame_rate:                            24

[led_effect second_segment_rgb]
leds:
    neopixel:awesome_led_name (9-16)
layers: 
    gradient       1 1  top  (1.0, 0.0, 0.0),(0.0, 1.0, 0.0),(0.0, 0.0, 1.0)
autostart:                             false
frame_rate:                            24

[led_effect third_segment_rgb]
leds:
    neopixel:awesome_led_name (17-24)
layers: 
    gradient       1 1  top  (1.0, 0.0, 0.0),(0.0, 1.0, 0.0),(0.0, 0.0, 1.0)
autostart:                             false
frame_rate:                            24

[led_effect fourth_segment_rgb]
leds:
    neopixel:awesome_led_name (25-32)
layers: 
    gradient       1 1  top  (1.0, 0.0, 0.0),(0.0, 1.0, 0.0),(0.0, 0.0, 1.0)
autostart:                             false
frame_rate:                            24

[led_effect all_segment_rgb]
leds:
    neopixel:awesome_led_name
layers: 
    gradient       1 1  top  (1.0, 0.0, 0.0),(0.0, 1.0, 0.0),(0.0, 0.0, 1.0)
autostart:                             false
frame_rate:                            24
```
Then start with:
```SET_LED_EFFECT effect=first_segment_rgb```

andreklug, 2022-11-03:
The, the setup works, but after restarting Klipper I can set a new effect only once through the Console (SET_LED_EFFECT). whatever I try afterwards doesn't have any effect. Here's my current config, I am trying to switch between the first three effects. changed the blending between top and bottom to see if that does something.

`[neopixel leds]
pin:                     PB0
chain_count:             144
color_order:             GRB
initial_RED:             0.1
initial_GREEN:           0.1
initial_BLUE:            0.1

[led_effect lights_on]
#autostart:              true
frame_rate:             60
leds:
    neopixel:leds (1-43)
    neopixel:leds (44-72)
    neopixel:leds (73-115)
layers:
    static 1 0 bottom (1,1,1)
#    breathing  10 1 top (.5,.5,1)

[led_effect lights_off]
#autostart:              true
frame_rate:             60
leds:
    neopixel:leds (1-43)
    neopixel:leds (44-72)
    neopixel:leds (73-115)
layers:
    static 1 0 bottom (0,0,0)
    
[led_effect lights_dim]
frame_rate:             60
leds:
    neopixel:leds (1-43)
    neopixel:leds (44-72)
    neopixel:leds (73-115)
layers:
 #   static 1 0 top (0.2,0.2,0.2)
    breathing  10 1 bottom (.5,.5,1)

[led_effect heater_progress]
leds:
    neopixel:leds (10-1)
autostart:                          false
frame_rate:                         60
heater:                             heater_bed
layers:
    heater  50 0 add    (1,1,0),(1,0,0)
    static  0  0 top    (1,0,0)
   
[led_effect critical_error]
leds:
    neopixel:leds (1-43)
layers:
    strobe         1  1.5   add        (1.0,  1.0, 1.0)
    breathing      2  0     difference (0.95, 0.0, 0.0)
    static         1  0     top        (1.0,  0.0, 0.0)
autostart:                             false
frame_rate:                            60
run_on_error:                          true

[led_effect panic]
leds:
    neopixel:leds (1-43)
layers:
    strobe         1  1.5   add        (1.0,  1.0, 1.0)
    breathing      2  0     difference (0.95, 0.0, 0.0)
    static         1  0     top        (1.0,  0.0, 0.0)
autostart:                             false
frame_rate:                            60

[led_effect progress_bar]
leds:
    neopixel:leds (10-1)
autostart:                          false
frame_rate:                         60
layers:
    progress  -1  0 add         ( 0, 0,   1),( 0, 0.1, 0.6)
    static     0  0 top         ( 0, 0, 0.1)`

gregspruce, 2022-11-05:
Is there a way to reverse the order on effects like chase and comet?

I tried  below listing all the light indexes in reverse instead of (9-16), and it seems to work for the heaterfire effect, but not the comet effect. I suppose I could terminate new wiring and physically flip my LED bars, but maybe I'm missing a tweak:

 ```
[led_effect bar1_meshing]
autostart:              false
frame_rate:             24
leds:
    neopixel:bar_leds (1-8)
layers:
    comet  3 1 top (1.0, 1.0, 0.2), (0.3, 0.3, 1.0), (0.0, 0.0, 1.0)

[led_effect bar2_meshing]
autostart:              false
frame_rate:             24
leds:
    neopixel:bar_leds (16,15,14,13,12,11,10,9)
layers:
    comet  -3 1 top (1.0, 1.0, 0.2), (0.3, 0.3, 1.0), (0.0, 0.0, 1.0)
 ```

julianschill, 2022-11-05:
Using a negative effect rate inverts the direction. Reversing the indexes also inverts the direction. So you inverted it twice which results in the original direction

julianschill, 2023-01-28:
As there don't seem to be anymore questions I will close this.

closed 2023-01-28 by julianschill