julianschill/klipper-led_effect #113: This specific config crashes the latest klipper version

Nyegaard, 2022-11-19:
I did a little experiment that didn't work but still had this config section in my led_effects.cfg file:
`#####################
## whole_top_strip ##
#####################

[led_effect top_strip_stroboscope]
leds:
    neopixel:top_strip
autostart:                          false
frame_rate:                         60
layers:
    strobe         110  100   top        (0.0,  0.0, 0.0, 1.0)

[gcode_macro stroboscope]
gcode:
    STOP_LED_EFFECTS
    SET_LED_EFFECT EFFECT=top_strip_stroboscope`

when starting klipper after I updated tonight, I got this error message. This was only happening on the chain connected to my klipper expander, so I tried upgrading the mcu there and on the octopus for good measure to all be on the latest klipper version, but no luck until I figured out it was this led effect that caused it, giving the following error in klippy.log:

`Unhandled exception during run
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 217, in run
    self.reactor.run()
  File "/home/pi/klipper/klippy/reactor.py", line 292, in run
    g_next.switch()
  File "/home/pi/klipper/klippy/reactor.py", line 340, in _dispatch_loop
    timeout = self._check_timers(eventtime, busy)
  File "/home/pi/klipper/klippy/reactor.py", line 158, in _check_timers
    t.waketime = waketime = t.callback(eventtime)
  File "/home/pi/klipper/klippy/extras/led_effect.py", line 177, in _getFrames
    frames = [(effect, effect.getFrame(eventtime)) for effect in self.effects]
  File "/home/pi/klipper/klippy/extras/led_effect.py", line 177, in <listcomp>
    frames = [(effect, effect.getFrame(eventtime)) for effect in self.effects]
  File "/home/pi/klipper/klippy/extras/led_effect.py", line 413, in getFrame
    layerFrame = layer.nextFrame(eventtime)
  File "/home/pi/klipper/klippy/extras/led_effect.py", line 485, in nextFrame
    return self.thisFrame[self.frameNumber]
IndexError: list index out of range
Transition to shutdown state: Unhandled exception during run`
[klippy (10).log](https://github.com/julianschill/klipper-led_effect/files/10048103/klippy.10.log)

wildfang, 2022-12-10:
I have the same problem unfortunately. Any fix in sight?

julianschill, 2022-12-11:
For the strobe effect the effect rate (the first parameter) can't be greater than the frame rate. So set it to 1 or so and it should work:
```strobe 1.00 100.00 top (0.0, 0.0, 0.0, 1.0) ```

I will add a check to the code.

wildfang, 2022-12-11:
It works when removing the strobe effect, thanks.

WesMcGee68, 2023-01-11:
I'm new to Klipper but have been dealing with a few issues when installing add-ons to it.  Most of them are due to the changes made to moonraker and klipper lately.  I noticed that in the install for this bit of software that it references the klipper_config folder.  That folder no longer exists in the newer installs of klipper.  Is this being addressed as well?

ayesim, 2023-01-11:
i am waiting to at least folder correction.

julianschill, 2023-02-04:
Fixed by v0.0.10

closed 2023-02-04 by julianschill