julianschill/klipper-led_effect #122: LED efect simulator GRB for BARf Voron led

locki-cz, 2022-12-26:
Simulator generate colours in RGB but voron 8 GRB leds ... it is possbile to make some switch for this?

julianschill, 2022-12-26:
You define the color order in the neopixel section. If you have mixed them in one chain (GRB for Barf and GRBW for nozzle) define it like this:
```color_order: GRB, GRB, GRB, GRB, GRB, GRB, GRB, GRB, GRBW, GRBW```

locki-cz, 2022-12-26:
I have it corrcetly in config, but it generate colours in simulator in RGB and I must rewrite it for every generated macro, I explain it probably wrong :)

julianschill, 2022-12-26:
All effects are always in RGB and then get sent in the correct order to the LEDs following the configured color order.

julianschill, 2023-01-28:
I will close this. There is nothing to be done here.

closed 2023-01-28 by julianschill