julianschill/klipper-led_effect #124: Heater_bed and extruder color gradient help

supercrossed, 2023-01-09:
Hi all, I have been trying to add an effect to cycle from blue>red with the bed and extruder. I'd like it to go blue>purple with the bed, and purple>red with the extruder.

I can do both effects separately, but can't figure out how to combine them. Below is my config for the LED strip.
[image]

julianschill, 2023-01-28:
Not sure if this is possible and also not sure what you want to achieve. Are the heaters heating one after the other or simultanously? How should the color change? LEDs can only show one color at ta time.

julianschill, 2024-04-28:
I think you should keep them as separate effects. Nothing wrong with that. Just make sure they don't have the same name as in your screenshot.

closed 2024-04-28 by julianschill