julianschill/klipper-led_effect #127: Issue: Klipper MCU freezes during progress effect

EmanuelStassar, 2023-01-20:
[Led effects config and printer config.zip](https://github.com/julianschill/klipper-led_effect/files/10467969/Led.effects.config.and.printer.config.zip)

As soon as printing progress starts, klipper crashes with an MCU error
[image]

A simpel progress effect works fine.

julianschill, 2023-01-20:
Ok, I will look into it. If you could also provide a klippy.log, that would help a lot.

EmanuelStassar, 2023-01-20:
[klippy-3.log](https://github.com/julianschill/klipper-led_effect/files/10470192/klippy-3.log)

Here is the Slippy log. And the version of led effects is: led_effect v0.0.9-0-g5d16b1c

I hope you can find the issue. BTW I managed to crash the simulator a few times while trying to get this effect right (on MAC OS on M1 MacBook Air)

julianschill, 2023-01-28:
As the error says it has "Lost communication with the MCU" I don't think that is error is caused by the effect. Maybe the LEDs draw too much power and your board shuts down? Hard to say. The exact same effect works on my printer.

EmanuelStassar, 2023-02-01:
That was it! Silly of me not to think about that. I reduced brightness of the white color and that solved the issue. I now have a nice green dot moving along a white ledstrip during printing. Thanks for looking into this and for the nice piece of coding!

closed 2023-02-01 by EmanuelStassar