julianschill/klipper-led_effect #128: Klipper hesitates during bed probing with high (>10) frame rates

NameOfTheDragon, 2023-01-22:
During Quad Gantry Levelling specifically, if there is an LED effect playing with a frame rate above about 10 frames per second, then the probing stutters and hesitates, sometimes for several seconds, and sometimes the levelling process simply stops until the firmware is restarted.

With a frame rate of 20 fps, it is about 50/50 whether QGL will complete. The impact lessens as the frame rate is reduced until a point is reached where the problem is not noticeable. For me on my Voron using an Octopus board, this is around 10 frames per second. I am using an animation very similar to the "critical error" animation.

Note: might be related to #127 .

julianschill, 2023-01-22:
Thanks for reporting this. Can you maybe provide a log? How is the Octopus connected to the pi?

NameOfTheDragon, 2023-01-22:
> Thanks for reporting this. Can you maybe provide a log? How is the Octopus connected to the pi?

Here's a video of the problem happening... https://youtu.be/vhIlNUq_0p0

The Octopus is connected to the Pi via USB. Log attached as requested.

[LED-Effects-Iss128-klippy.log](https://github.com/julianschill/klipper-led_effect/files/10475772/LED-Effects-Iss128-klippy.log)

julianschill, 2023-01-28:
As this is not happening for me and you have also the other issue, I suspect that there is something wrong in your setup. Maybe try changing the USB cable to your MCU. Also check with `top` or `htop` or in mainsail if there is any high load on your pi. You can also check the load on the MCU.

julianschill, 2023-01-28:
One solution would be to move the LEDs to a separate MCU. That would take the load off the Octopus.

NameOfTheDragon, 2023-12-15:
My printer died due to cabling issues and has been down for about 6 months, just resurrected it. I've replaced my 2xSKR boards with an Octopus board. This is still happening for me.
I did winder if I might be overloading the controller as my LEDs can draw about 8 amps worst-case, so I have installed a separate 5V supply and moved my LEDs over onto that. No change, unfortunately.
There is not high load on the system. Mainsail reports a maximum of about 14% while printing and the controller load is a constant zero, which might mean its not reporting the correct value but the "awake time" is changing between 0.11 and 0.12 so something is working there.

Do you really think this can be a USB cable issue, given that the printer is printing correctly? I seriously doubt it.

I'm at a loss to explain why this is happening, there doesn't seem to be high load at all, but the problem completely goes away if I disable LED effects during QGL.

NameOfTheDragon, 2025-02-01:
FYI upgrading to a Raspberry Pi 4 has solved all my issues. Some people have suggested that this problem is worse on Raspberry Pi 3 with 64-bit Raspian OS. That was indeed what I was running so I upgraded to Raspberry Pi 4 and it is a night-anmd-day difference. I was motivated to upgrade when I installed an ERCF MMU and the Happy Hare firmware caused a lot of shutdowns. So anyone having this issue, get yourself a Raspberry Pi 4 (now that the 5 is out, the price should drop!).