julianschill/klipper-led_effect #132: Section 'led_effect idle' is not a valid config section

Doodys, 2023-02-11:
I've been using led_effect for a long time and after updating klipper (this big update to cooperate with new directories in the new moonraker) my printer cannot use any `led_effect` section. 

I'm updated to the `led_effect v0.0.10-0-g35cf587`, everything is stored in the separate cfg file which is included in the printer.cfg. When I fix those values in the printer.cfg instead of the separate leds.cfg file it still cannot read `led_effect xxx` sections.

I did not change anything in `led_effect` sections and double checked if there are no new syntaxes for them after any update - everything looks right. 

Is there any known solution for this? My `WS2812B_leds.cfg`:
[WS2812B_leds.txt](https://github.com/julianschill/klipper-led_effect/files/10713310/WS2812B_leds.txt)

And `klippy.log`:
[klippy.log](https://github.com/julianschill/klipper-led_effect/files/10713315/klippy.log)

Any help would be appreciated. Thanks!

julianschill, 2023-02-11:
Probably the link to the plugin file was removed during the update. Just run the install script again.

Doodys, 2023-02-12:
Yeah, for some reason link to the py file in the klippy folder wasn't working as expected - standard `turn it off and on` (or rather `uninstall and install`) helped in this case. Thanks.

closed 2023-02-12 by Doodys

thijstriemstra, 2023-03-27:
> Probably the link to the plugin file was removed during the update. Just run the install script again.

I'm running into the same issue, is there any workaround for this? Can this ticket be re-opened (e.g. this extension gets uninstalled when doing a klipper update)?

julianschill, 2023-03-27:
Do firstly this is not an issue of this plugin, but of the moonraker updater. Secondly klipper has no official plugin support and what we here do is actually a hack.
There is nothing I can do about it. Just run the installation script after an update and you should be fine.