julianschill/klipper-led_effect #133: Heater effect not fading

danicedenv, 2023-02-22:
[image]
Hey everyone so I'm having issues with the heater effect. I don't know if I'm doing something wrong but instead of fading it simply swaps to the second colour as soon as the temperature is reached. Thanks in advance.

julianschill, 2023-02-22:
The last color is just used to switch to when the temperature is reached. So you have only defined one color to fade through. Add another color.

danicedenv, 2023-02-22:
I get it thanks. I think what confused me is that on the doc it says "Once it has been met, the last color of the gradient..." maybe it would be better to say the last colour of the palette. Brilliant work so far, thanks for all this.

julianschill, 2023-03-01:
You are right. I will either change the behaviour of the heater effect or improve the wording in the documentation.