julianschill/klipper-led_effect #139: Install on multi instance klipper setup

jakobwowy, 2023-03-29:
I tried to install it on my rapi where I have 4 klipper instances running. 
The install script failed because no klipper instance was found:
"Klipper service not found, please install Klipper first"

The naming of the services is:

klipper-1.service                                           loaded    active     running       Klipper 3D Printer Firmware SV1
klipper-2.service                                           loaded    active     running       Klipper 3D Printer Firmware SV1
klipper-3.service                                           loaded    active     running       Klipper 3D Printer Firmware SV1
klipper-4.service                                           loaded    active     running       Klipper 3D Printer Firmware SV1

julianschill, 2023-04-16:
Just put a softlink to led-effect.py in the klipper/extras directories. You can also add an updater section to your moonraker.conf. That's all the script does.

cdw2929, 2023-08-27:
would you mind giving a hint as what to do
googling how to make a softlink and trying to figure out where the location of both the led-effect.py and the klipper/extras location would the location be klipper/klippy/extras? if so I have plenty of .py files but not one led-effect.py

I think the problem is is that if you have multiple instances on one machine it doesnt install

I renamed the if condition in the install script to find the service 
but when it installs and tries to start the service it doesnt use the amended name

the installer stops klipper and tries to restart it but cant because the services instances are named different than just klipper.service

how do we change that

I sure this is an easy fix for someone that knows the language 
if you dont mind showing some of us that dont so we dont have to spend days googling

julianschill, 2023-09-07:
run
```
cd klipper-led_effect
ln -s led_effect.py ~/klipper/klippy/extras
```

This will create the link.

Then add this to the moonraker.conf of one of your instances for the updater:

```
[update_manager led_effect]
type: git_repo
path: ~/klipper-led_effect
origin: https://github.com/julianschill/klipper-led_effect.git
 ```

julianschill, 2024-05-27:
I added the manual installation method to the README. Also you can specify the name of the klipper service for the script with the -s parameter.

closed 2024-05-27 by julianschill