julianschill/klipper-led_effect #140: Effect not tracking progress

Tomadoggy, 2023-03-29:
Love the effects!  Thank you for all of your efforts!

I have a properly configured 30 LED strip:

[neopixel front_strip]
pin: PC6                                 # RGB1
chain_count:30                           # Count of LED's wired in series within strip
color_order: GRB                         # If RGBW, enable initial_white below
initial_RED: 0.0
initial_GREEN: 0.0
initial_BLUE: 0.0

And have tested this effect in the simulator and it displays as I desire:

[led_effect LED_PRINT_PROGRESS] 
autostart:              false
frame_rate:             24
leds:
    neopixel:front_strip
layers:
  progress -1.00 0.00 add (0.00,1.00,0.00),(0,0.1,0.6) 
  static 0.00 0.00 top (0,0,0.1) 

However when I call the effect (at the end of my print start macro), the led strip lights up blue to LED #17 immediately rather than starting at LED #1, and the green top LED does not display.  At 100% completion the strip properly displays all the way up the strip with the green top LED showing.  Essentially goes from start at middle with no progress shown during print, then to 100%

Thank you in advance.

julianschill, 2023-04-16:
Maoe sure there are M73 commands in your gcode to set the progress.

julianschill, 2023-04-29:
you have to add [display_status] 
https://www.klipper3d.org/G-Codes.html#display_status

Tomadoggy, 2023-04-30:
Thank you very much!

closed 2023-09-18 by julianschill