julianschill/klipper-led_effect #141: led_effect_erreur

Eloura74, 2023-03-31:
[image]

[image]

Hello, I have a real problem and I don't understand at all why.
since the klipper update, no effect works anymore, is anyone in this case?

gdachs, 2023-03-31:
You just have not installed led_effects, or lost it on the update. [Install it again](https://github.com/julianschill/klipper-led_effect#installation)

Eloura74, 2023-03-31:
[image]
However, it is installed correctly, how do I uninstall it?

gdachs, 2023-03-31:
You didn't see the link?

Eloura74, 2023-03-31:
it was actually very simple, thank you very much :)

closed 2023-03-31 by Eloura74