julianschill/klipper-led_effect #143: Not lighting entire LED string

Peaches491, 2023-04-16:
I've recently switched from Marlin/Octoprint to Klipper/Mainsail and was very happy to find this library.

After copying over my settings directly from my Marlin config file, this implementation fails to light all of the LEDs in the strip. Previously, on Marlin the LEDs had been working great for years, so I don't suspect it has anything to do with the wiring or signal integrity. 

Do you have any ideas what might be causing this? It's only reliably lighting ~30 of the 70 LEDs in my chain 😞 

```
[neopixel panel_ring]
pin: PL5
chain_count: 69
initial_RED: 0.0
initial_GREEN: 0.0
initial_BLUE: 0.0
initial_WHITE: 1.0
color_order: GRBW
```

julianschill, 2023-04-16:
Are the colors correct? If not try changing to color order GRB. Can you set it without effects by just using SET_LED?

Peaches491, 2023-04-17:
They're RGBW LEDs, and I've confirmed that GRBW is the correct color order. 

Testing out `SET_LED` and I see the same behavior. Only some light up, and not very reliably further down the chain. It feels like something is off in the timing where it can control the first several LEDs reliably, but further out it gets flaky

julianschill, 2023-04-29:
If it's also not working with SET_LED, it's not a problem of this plugin. Make sure that it works with the Klipper built-in commands before trying effects.

treaves, 2024-01-17:
I had a similar issue with my Neopixel string. To fix my issue I had to edit the signal timeout value in one of the Klipper Python files. [Here is the thread](https://klipper.discourse.group/t/issues-with-45-led-neopixel-chain/4026/7) if you want to see if your issue is related b

closed 2024-04-28 by julianschill