julianschill/klipper-led_effect #145: Using variable

Belfaigore, 2023-04-22:
Hello,

Is there a way to use variable like `{printer.heater_bed.target}` in led_effect ; for example
```
[led_effect bed_temp]
; Description: Gradient led color in function of bed temperature
autostart:  false
frame_rate: 24
leds:
    neopixel:sb_leds (1-8)
heater:     heater_bed
layers:
    fire 20.00 8.00 multiply (0.0,0.0,0.0),(1.00,1.00,1.00)
    temperature  {printer.configfile.settings.heater_bed.min_temp} {printer.heater_bed.target} add  (0,0,0.1),(1,0,0)
```

julianschill, 2023-09-18:
Not at the moment. I will consider some functionality like this for a future update.

Benoitone, 2024-03-13:
Hello, It will be fine also to be able to use an user variable for led idx: for exemple:
```
[led_effect bed_temp]
; Description: Gradient led color in function of bed temperature
autostart:  false
frame_rate: 24
leds:
    neopixel:sb_leds ({printer['gcode_macro _USER_VARIABLES'].status_leds_logo_idx})
heater:     heater_bed
```

Paul-Damian, 2024-05-11:
Please do so :-)
Thank you for your work.

julianschill, 2024-05-18:
There is a new feature to use parameters for effects with template expansion. https://github.com/julianschill/klipper-led_effect/blob/master/docs/LED_Effect.md#template-processing

Maybe this solves already what you want to do.

Paul-Damian, 2024-05-21:
Thank you! It is exactly what I hoped for!