julianschill/klipper-led_effect #146: adding `led_effect sb_nozzle_printing` issue

Gil80, 2023-04-22:
When printing, I'd like the nozzle leds to be white, so the webcam can pick the printing process easily.

I added the following to the `stealthburner_led_effects_3_leds.cfg`:

Under "Nozzle effects"
```
[led_effect sb_nozzle_printing]
autostart:              false
frame_rate:             24
leds:
    neopixel:sb_leds (2,3)
layers:
        static 0  0 subtract (1.0, 1.0, 1.0)
```
and edited this:
```
[gcode_macro status_printing]
gcode:
    STOP_LED_EFFECTS
    SET_LED_EFFECT EFFECT=sb_logo_printing
    SET_LED_EFFECT EFFECT=sb_nozzle_printing
    set_nozzle_leds_on
```

The issue is that the nozzle leds are too bright when viewing the printing process through the webcam and I couldn't find how to set `WHITE=0.1` only to the nozzle leds.
With a previous `stealthburner_leds.cfg` config file I used, I could issue: `_set_nozzle_leds WHITE=0.1` and it would change it accordingly. 

Appreciate your help!

claudermilk, 2023-04-28:
I have my nozzle lights set with add, not subtract. I'm using the spec parts, so they are RGBW and my definition is static 0 0 add (0.8,0.8,0.8,1.0) which works. You can drop the levels from there if needed.

Gil80, 2023-05-03:
@claudermilk I couldn't figure out how to lower the LED output. What's the command to use?
Before this great LED config, I could send a command with WHITE=0.1 and all the nozzle leds were at their lowest output.

Gil80, 2023-05-04:
> 1.0 = 100%. To set to 10% for example, just use (0.1,0.1,0.1,0.1) The values are (R,G,B,W)

is there a command I can run via console to see the change?

claudermilk, 2023-05-07:
I like using the simulator. Then I just tweak the settings in my definitions, restart Klipper and run the macro to set the LEDs and see what they look like. Lots of restarting Klipper until you're happy.

Gil80, 2023-05-07:
@claudermilk thanks. I wasn't able to figure out how to use the simulator... I'll try again.

julianschill, 2023-05-07:
Go to releases, download the file for your platform (Windows or MacOS) and run it. There is also a video from West3d that explains how to use it.
https://youtu.be/ZRuwBEOEA2c

Gil80, 2023-05-07:
@julianschill I had a play with the simulator and watched the youtube video and I still don't understand how to get this working for my stealthburner.
I have the Voron official 3 LEDs chain. 1 for the logo, 2 for the nozzle.
All I want is to have the logo display whatever print gradient, like comet, but the nozzle LEDs to be the weakest **static** white light possible.
I wasn't able to achieve this with the simulator. I also read the documentation about the blending options, but it seems that it's beyond my understanding how to operate this.

Is there a simple macro to apply when a print starts, given the led nozzles to be static white with lowest light output?

julianschill, 2023-05-07:
Just modify your nozzle effect:
```
[led_effect sb_nozzle_printing]
autostart:              false
frame_rate:             24
leds:
    neopixel:sb_leds (2,3)
layers:
        static 0  0 top (0.1, 0.1, 0.1) 
```

closed 2023-09-18 by julianschill