julianschill/klipper-led_effect #147: Homing Effect

claudermilk, 2023-04-28:
The homing effect doesn't seem to be working. I'm not getting a response when an effect is just set to flash white on homing (currently SB with spec nozzle LEDs & Rainbow Barf logo). Using spec switch X & Y endstops with the PCB and Tap for Z.

julianschill, 2023-04-28:
Post the config of the effect

claudermilk, 2023-04-29:
homing 16 00 top (1.0,1.0,1.0)

Set to flash white and fade slowly just to see if it does anything. I'm pretty sure this is ok from the docs. I generated it from within the simulator. I have it combined with breathing, but I still get nothing even when eliminating all other effect layers.

julianschill, 2023-04-29:
The layer looks ok and works for me in the simulator. Please post the complete config of the effect.

julianschill, 2023-05-02:
You didn't specify which endstops to react on. See the documentation here: https://github.com/julianschill/klipper-led_effect/blob/master/docs/LED_Effect.md#additional-effect-level-parameters

Add the "endstops" parameter to the effect. 

Example:
`[led_effect nozzle_homing]
leds:
neopixel:sb_leds (9,10)
autostart: false
frame_rate: 24
endstops: x, y, probe
layers:
breathing 8 0 add (0,0.6,0.2),(0,0.46,0.15)
homing 16 00 top (1.0,1.0,1.0)`

claudermilk, 2023-05-03:
That works! I added endstops: x, y, z and it works as expected. probe did not work for Z. This isn't mentioned in the docs, so they need to be updated.

Currently in docs:
Homing
Effect Rate:  1   Determines decay rate. A higher number yields slower decay
Cutoff:       0   Not used, but must be provided
Palette:          Colors are cycled in order

julianschill, 2023-05-03:
It's explained further up in the "additional parameters" section. But I agree, documentation needs a rework. When using a virtual endstop for Z it will trigger on "z", when homing and on "probe" when meshing or QGLing.

closed 2023-05-03 by julianschill