julianschill/klipper-led_effect #156: Temperature_range

Silent4719, 2023-08-11:
Hello,

Would there be a simple solution to activate an effect according to a temperature range?

The ideal would be something like that: 
```

[led_effect stroboscope]
leds:
    neopixel:case (1-10)
autostart:                          false
frame_rate:                         10
heater:                             extruder
temperature_range:                  170-200
layers:
    strobe 0.1 0.5 top (1.0, 0.0, 0.0)
    static 0 0 subtract (0.9, 0.9, 0.9)
```

The stroboscopic and static effect would activate when the extruder is between 170 and 200 °.

julianschill, 2023-09-18:
Should be possible when you multiply a heater effect. Put (1,1,1,1) as the last color and multiply with the effect you want to see. It will switch to (1,1,1,1) when the temperature is reached.

No temperature range at the moment, though.

julianschill, 2024-05-18:
This should work:
```
strobe 1.00 0.50 multiply (1.00,0.00,0.00) 
static 0.00 0.00 divide (1.00,1.00,1.00) 
temperature 170 200 top (0.00,0.00,0.00),(1.00,1.00,1.00),(0.00,0.00,0.00) 
```